adafruit/Adafruit_Learning_System_Guides #2: Use pycodestyle to lint check all examples

tannewt, 2017-10-10:
It enforces pep8: https://github.com/PyCQA/pycodestyle

tannewt, 2018-05-14:
We do this in libraries already and should follow the example settings there: https://github.com/adafruit/cookiecutter-adafruit-circuitpython/blob/master/%7B%7B%20cookiecutter.library_name%20%7D%7D/.travis.yml#L30

craigargh, 2018-05-14:
I'm going to work on:
- simple_strand_palettes
- uf2_flash_erasers
- ufo

ladyada, 2018-05-14:
one possible route - have the pylint script live in the root of the repo, and run on any .py script it finds, UNLESS a .circuitpython.skip file exists in the same directory of the .py script

dylwhich, 2018-05-14:
I'm working on adding a pylint script and a travis-ci config to run it

craigargh, 2018-05-14:
@dylwhich Would you be able to add a pylint config file as well?

ajdershem, 2018-05-14:
I am working on:

- 3D_Printed_NeoPixel_Gas_Mask

craigargh, 2018-05-14:
I'm using the following to run pylint
```
pylint --disable=missing-docstring,invalid-name,bad-whitespace,duplicate-code,import-error,consider-using-enumerate,no-name-in-module,no-member,too-many-arguments,no-else-return */*.py
```

It might make sense to Flake8 instead of PyLint as it's less opinionated.

kattni, 2018-05-16:
@craigargh `@staticmethod` is implemented in CircuitPython.

dhalbert, 2018-07-07:
Fixed by #166. Thanks @craigargh !

closed 2018-07-07 by dhalbert