adafruit/Adafruit_Learning_System_Guides #53: Suggestion for change for reading PMS5003 packet in Python

krcurtis, 2018-02-20:
I ran into a problem where the struct.unpack() was failing on the BeagleBone, (due to how I changed the example to flush the serial buffer, and take measurements at a different interval) and the problem seemed to be that the bytearray object was too long for the format ( ">HH..."). That could be a MicroPython vs Python 2.7 issue, I'm not sure. In any case I ended up with a revised PMS5003 packet reading/parsing function to contribute:

```python
def read_valid_packet_from(ser):
    buffer = bytearray()
    valid_packet = False

    while not valid_packet:
        n = min(32, 32- len(buffer))
        data = ser.read(n)
        data = bytearray(data)
        print("read: ", data)          # this is a bytearray type 
        buffer += data

        # look for packet start
        while buffer:
            if buffer[0] == 0x42:
                if len(buffer) >= 2:
                    if buffer[1] == 0x4d:
                        # found valid packet start
                        break
                    else:
                        buffer = buffer[1:] # re-start parsing on next byte
                        continue
                else:
                    break # might be valid packet start, not enough data
            else:
                buffer = buffer[1:] # re-start parsing on next byte
                continue

        #print("current buffer_len", len(buffer))
        #print("current buffer: ", buffer)

        if len(buffer) < 32:
            print("buffer_len", len(buffer))
            print("buffer: ", buffer)
            continue

        frame_len = struct.unpack(">H", bytes(buffer[2:4]))[0]
        print("frame length", frame_len)
        if frame_len != 28:
            # packet is invalid, yet had valid start pattern
            buffer = buffer[2:] # restart parsing after intitial start pattern
            continue

        checksum = struct.unpack(">H", bytes(buffer[30:]))[0]

        check = sum(buffer[0:30])
        if check != checksum:
            print("checksum didn't match", check, checksum)
            buffer = bytearray() # drop whole packet
            continue
        else:
            print("checksum passed")
            valid_packet = True

    return buffer
```

ladyada, 2018-04-25:
thanks! but this still requires unpacking all the data no? did you try running the exact example code, as unpack() goes, its not a _huge_ string :)

ladyada, 2018-04-25:
ok the latest commits do some more flushing-if-invalid checks, please try it :)

closed 2018-04-25 by ladyada