adafruit/Adafruit_Learning_System_Guides #338: Crickit feather neopixel missing brightness

mnelsoneorm, 2018-09-22:
I've noticed that the brightness setting doesn't work on the neopixel library on Crickit Feather.  It does work on the circuitplayground.

ladyada, 2018-09-22:
dont forget you hvae to set the brightness first! if you still have the bug, open up an issue here with example code that you know doesn't work 
thx!
https://github.com/adafruit/Adafruit_Seesaw

closed 2018-09-22 by ladyada

mnelsoneorm, 2018-09-22:
feather M4 on crickit.  Circuitpython.

`import time
from adafruit_crickit import crickit
from adafruit_seesaw.neopixel import NeoPixel

num_pixels = 20

pixels - NeoPixel(crickit.seesaw, 20, num_pixels, brightness=0.2)`

Whatever I set the brightness to it runs at full brightness.  The playground express on the crickit does work just fine.  I'm following the guide "Make It Glow With Crickit".  The guide also has brightness set for the circuit playground example but not on the feather example.

ladyada, 2018-09-22:
ok please open the issue on the library linked above!