adafruit/Adafruit_Learning_System_Guides #456: a lot of errors when trying to compile for Feather M0 Express

Impact9, 2018-12-30:
Adafruit is unclear with exactly how to perform this project. I'm trying to compile this for my Adafruit Feather M0 Express Hallowing but I'm finding a lot of errors and being a novice I'm not sure what's wrong, me or the code. Following this guide (https://learn.adafruit.com/hallowing-all-seeing-skull/code-in-arduino) to here. 

I did load the libraries and boards instructed but the only thing that wasn't clear was the Programmer. I've tried AVRISP MK II with the following results on Arduino 1.8.8 - 

`Arduino: 1.8.8 (Windows 10), Board: "Adafruit Feather M0 Express"

All_Seeing_Skull:290:7: error: stray '\342' in program

       <span aria-hidden="true" class="d-inline-block ml-1 v-align-middle">â†µ</span>

       ^

All_Seeing_Skull:290:7: error: stray '\206' in program

All_Seeing_Skull:290:7: error: stray '\265' in program

All_Seeing_Skull:1621:70: error: stray '#' in program

         <td id="LC21" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">include</span> <span class="pl-s"><span class="pl-pds">&lt;</span>SPI.h<span class="pl-pds">&gt;</span></span></td>

                                                                      ^

All_Seeing_Skull:1625:70: error: stray '#' in program

         <td id="LC22" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">include</span> <span class="pl-s"><span class="pl-pds">&lt;</span>Adafruit_GFX.h<span class="pl-pds">&gt;</span></span></td>

                                                                      ^

All_Seeing_Skull:1629:70: error: stray '#' in program

         <td id="LC23" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">ifdef</span> ARDUINO_ARCH_SAMD</td>

                                                                      ^

All_Seeing_Skull:1633:72: error: stray '#' in program

         <td id="LC24" class="blob-code blob-code-inner js-file-line">  #<span class="pl-k">include</span> <span class="pl-s"><span class="pl-pds">&lt;</span>Adafruit_ZeroDMA.h<span class="pl-pds">&gt;</span></span></td>

                                                                        ^

All_Seeing_Skull:1637:70: error: stray '#' in program

         <td id="LC25" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">endif</span></td>

                                                                      ^

All_Seeing_Skull:1688:70: error: stray '#' in program

         <td id="LC37" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">include</span> <span class="pl-s"><span class="pl-pds">&quot;</span>config.h<span class="pl-pds">&quot;</span></span>     <span class="pl-c"><span class="pl-c">//</span> ****** CONFIGURATION IS DONE IN HERE ******</span></td>

                                                                      ^

All_Seeing_Skull:1697:70: error: stray '#' in program

         <td id="LC39" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">if</span> defined(_ADAFRUIT_ST7735H_) || defined(_ADAFRUIT_ST77XXH_)</td>

                                                                      ^

All_Seeing_Skull:1705:70: error: stray '#' in program

         <td id="LC41" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">else</span></td>

                                                                      ^

All_Seeing_Skull:1713:70: error: stray '#' in program

         <td id="LC43" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">endif</span></td>

                                                                      ^

[... 768 lines cut ...]
All_Seeing_Skull:1838:215: error: expected unqualified-id before '<' token

         <td id="LC73" class="blob-code blob-code-inner js-file-line">  <span class="pl-k">static</span> <span class="pl-k">volatile</span> <span class="pl-k">bool</span> dma_busy = <span class="pl-c1">false</span>;</td>

                                                                                                                                                                                                                       ^

All_Seeing_Skull:1842:247: error: expected unqualified-id before '<' token

         <td id="LC74" class="blob-code blob-code-inner js-file-line">  <span class="pl-k">static</span> <span class="pl-k">void</span> <span class="pl-en">dma_callback</span>(Adafruit_ZeroDMA *dma) { dma_busy = <span class="pl-c1">false</span>; }</td>

                                                                                                                                                                                                                                                       ^

All_Seeing_Skull:1855:118: error: expected unqualified-id before '<' token

         <td id="LC77" class="blob-code blob-code-inner js-file-line"><span class="pl-c1">uint32_t</span> startTime;  <span class="pl-c"><span class="pl-c">//</span> For FPS indicator</span></td>

                                                                                                                      ^

exit status 1
stray '\342' in program

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

ladyada, 2018-12-30:
hi please go back to the giude and click this button, you shoudl get a zip

[image]

closed 2018-12-30 by ladyada

Impact9, 2018-12-31:
I'm not sure why but it compiled and uploaded this time but it **will not** work. It did initially work when I first powered it up but I was wanting to change eyes which caused this problem. Now I did have to delete the Arduino15 folders that were hidden in my user folder to get this to compile. I have both the Arduino and Adafruit SAMD along with the Adafruit GFX, ST7735/ST7789, and Zero DMA included. I just can't get it to work. I shouldn't need to do anything else beyond commenting out the OLED lines in the config.ch right?

ladyada, 2018-12-31:
if you have other tech support questions please post in the forums - we dont do tech support in github issues :)

Impact9, 2018-12-31:
Understandable but it's a code issue related to this "project" not a hardware issue.