adafruit/Adafruit_Learning_System_Guides #527: Rack Pinion Project - Invalid f3d and step files

basktrading, 2019-02-23:
Hi, it seems that the source f3d and step files in the Rack Pinion Project are invalid, can you please check? Thanks!

djecken, 2019-02-23:
Sure, I download from the repo and uploaded to latest version of Fusion 360. Both files check out. Is there a specific error? Try using "Save Link As" option when left-clicking download button.

basktrading, 2019-02-24:
Thanks for your quick response ,,  you are correct, my bad. Relatively new to github and I wasn't correctly 
downloading the file as I usually would using right click on link, then save as. Sorry for the bother, have a super weekend!

closed 2019-02-24 by basktrading

webseed50, 2021-04-14:
Where can i download the rack and pinion f3d file?

djecken, 2021-04-15:
@webseed50 try downloading from the github repo or thingiverse. github link: https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/master/Rack_Pinion_Bot/source thingiverse link: https://www.thingiverse.com/thing:3038784/files