adafruit/Adafruit_Learning_System_Guides #579: alarm clock snooze crash

djecken, 2019-03-25:
@dastels Pressing the snooze button while in clock is in"snoozing mode" causes the code to crash. Also happens when going into the settings page while the clock is snoozing.

[image]

djecken, 2019-03-26:
@dastels tested latest code. getting there! 

- the runtime error occurs on the 4th snooze event.
- after the alarm background loads the sound is delayed by 3-secs.

dastels, 2019-03-26:
Ok, I’ll look into the error. Same as before? The sound delay is just how the timing is done. I’ll change that.

Dave

djecken, 2019-03-26:
yea same "group full" type. Think it would sound nice to play this 10sec looping song. 
[song-mix.zip](https://github.com/adafruit/Adafruit_Learning_System_Guides/files/3006019/song-mix.zip)

dastels, 2019-03-26:
Fix is being tested... I'm at 7 or 8 snooze events and it's solid.
I've changed the alarm sound playing so that it plays "alarm.wav" so you can drop in whatever file you want. The file you provided played a couple seconds and started just ticking (on the PyPortal).  It played fine in Linux.
FYI I'm using 4.0 beta 3 and 5 (and head) still are too buggy.

djecken, 2019-03-26:
nice! i'm using 4.0 beta 5 as well. I also tested the audio file and got similar "clicking" behavior.

dastels, 2019-03-26:
My phrasing was ambiguous. I'm using Beat 3. I tried Beta 5 and found it not stable enough. I'm using the latest (yesterday) bundle release, though.

dastels, 2019-03-26:
Updated PR. I went through a dozen or so snooze events without any issues.

djecken, 2019-03-26:
switched over to beta 3 and still getting a "clicking" noise with the 10-sec wav file. any suggestions? snooze seems solid now.

djecken, 2019-03-26:
OK I trimmed the wav file down to ~2seconds and reduced to 22.05kHz and still getting clicks - it won't play the whole sound.

djecken, 2019-03-26:
OK trying a different audio file. this time it's a wav 16-bit interger, 16kHz, mono. Doesn't play more than a 1/4 sec of the sound followed by the clicking. 
[alarm.zip](https://github.com/adafruit/Adafruit_Learning_System_Guides/files/3008730/alarm.zip)

djecken, 2019-03-26:
@dastels can you download the newer audio file and test to see if the issues persists? thanks!

dastels, 2019-03-26:
"Blip.. tick tick tick"

djecken, 2019-03-26:
Ahh hrmmm. I've tried three different audio files now and can't get any of them to work like the original alarm sound you have. Any suggestions?

dastels, 2019-03-27:
This last one has the same sample rate & bitrate as the original I used, so I'm stumped by it. Duration and file size almost the same..

I reconverted/exported it and get the same thing "blip tick tick tick"

djecken, 2019-03-27:
strange! i wonder if this could this be a caching issue?

dastels, 2019-03-27:
I'm not familiar enough with the sound player internals to say.

ladyada, 2019-04-02:
hiya folkx is this resolved?

closed 2019-04-03 by djecken