adafruit/Adafruit_Learning_System_Guides #751: PyPaint Errors

djecken, 2019-07-09:
@dastels Hiya, I'm testing this out. Running PyPaint on PyGamer using circuit python 4.0.2 with latest libraries from bundle.

- Analog joystick is inverted when moving cursor
- Pressing A button triggers error, exits program and loads console on screen.
- From console: cursorcontrol.py, Line 177, "group" object has no attribute "remove"

djecken, 2019-07-09:
Just tested **PyBadge** with circuit python **4.0.2** getting same errors on **Line 177,** in hidden attributeError: 'group' object has no attribute 'remove'.

tannewt, 2019-07-09:
Please try it with the Beta. I believe that is when remove was added.

dastels, 2019-07-09:
As Scott said, use the latest beta.

djecken, 2019-07-10:
confirmed works with 4.1.0 beta, yay!

closed 2019-07-10 by djecken