adafruit/Adafruit_Learning_System_Guides #798: current PyPortal_ArduinoSelfTest using deprecated library and does not work even with it

mMerlin, 2019-08-15:
`PyPortal_ArduinoSelfTest/PyPortal_ArduinoSelfTest.ino` includes "Adafruit_QSPI_Flash.h", which is not available in the Arduino IDE library manager.  https://learn.adafruit.com/adafruit-pyportal/arduino-libraries shows installing library `Adafruit QSPI`.  Online searching found https://github.com/adafruit/Adafruit_QSPI which says it is deprecated and no longer needed.

Commenting out the include, definition and block that references flash got it to run.  Initial exploration with manual installation of the deprecated QSPI library, SdFat, or `SdFat - Adafruit Fork' failed various ways too.

PyPortal_ArduinoSelfTest.ino needs some attention, as well as «at least» the web pages [https://learn.adafruit.com/adafruit-pyportal/arduino-libraries](https://learn.adafruit.com/adafruit-pyportal/arduino-libraries) and [https://learn.adafruit.com/adafruit-pyportal/arduino-test](https://learn.adafruit.com/adafruit-pyportal/arduino-test) that reference the selftest program and needed libraries.

Environment:
```txt
PyPortal
Fedora 30: 5.2.7-200.fc30.x86_64
Arduino IDE 1.8.5
flatpak info cc.arduino.arduinoide

Arduino IDE - Open-source electronics prototyping platform

          ID: cc.arduino.arduinoide
         Ref: app/cc.arduino.arduinoide/x86_64/stable
        Arch: x86_64
      Branch: stable
     Version: 1.8.5
     License: LGPL-2.1
      Origin: flathub
  Collection: org.flathub.Stable
Installation: user
   Installed: 445.4 MB
     Runtime: org.freedesktop.Platform/x86_64/18.08
         Sdk: org.freedesktop.Sdk/x86_64/18.08

      Commit: defc1c965961c09ccf1a2eb2deca8b5645b83266cb77eb80e7b682925b8ff382
      Parent: 920a50ca67d0e64cb70ed32311fe5d674e3cbc6ebfe82b431e96fa6536769c0f
     Subject: Update runtime to version 18.08 (0242af5a)
        Date: 2018-11-08 15:20:46 +0000

Adafruit ADT7410 Library by Adafruit Version 1.0.1
Adafruit GFX Library by Adafruit Version 1.5.6
Adafruit ILI9341 by Adafruit Version 1.5.1
Adafruit ImageReader Library by Adafruit Version 2.0.1
Adafruit NeoPixel by Adafruit Version 1.2.4
Adafruit QSPI by Adafruit Version 3.0.0
Adafruit SPIFlash by Adafruit Version 3.1.1
Adafruit TouchScreen by Adafruit Version 1.0.2
Adafruit Zero DMA Library Built-In by Adafruit Version 1.0.4
SdFat - Adafruit Fork by Bill Greiman Version 1.2.1
SdFat by Bill Greiman Version 1.1.0
```

ladyada, 2019-08-15:
try https://github.com/adafruit/Adafruit_Learning_System_Guides/pull/789

mMerlin, 2019-08-15:
That works!  Now just need to get that merged, plus the web pages and zip download updated.

Might want to do a small tweak first.  The display shows:

```txt
QSPI Flash...QSPI Fl
ash JEDEC 0x…
```
May want a line break, or a line break and remove the duplicate `QSPI Flash` at line 85
Or put the line break back at line 75

```c++
  tft.println("QSPI Flash...");
…
  tft.print("JEDEC 0x"); tft.println(flash.getJEDECID(), HEX);
```

ladyada, 2019-08-15:
thanks i merged it - will fix typos later, it works well enuf :)

closed 2019-08-15 by ladyada

mMerlin, 2019-08-15:
Also just noticed, while looking how to turn down the backlight brightness, that you have 

```c++
#define TFT_BACKLIGHT 25
```

in the code twice.  Lines 16 and 27
You have 2 names for the tft reset pin.  Both being used.
```c++
#define TFT_RESET     24
#define TFT_RST       24 // Reset pin
```
`TFT_RST` is only used one place, in the call to the constructor.

Even changing TFT_BACKLIGHT to analogWrite at 10, the temperature sensor reads pretty hot.  At full on, it was about 30°C.  At pwm of 10, it is down to about 27.7°.  Ambient is around 21.  Turning it off for awhile got it down to 25°, but even with setting pwm 1, it started to climb again.  Have to see if I can calibrate a constant offset based on the pwm setting.