adafruit/Adafruit_Learning_System_Guides #846: Slideshow Soundtrack Error

djecken, 2019-09-19:
Testing **slideshows_slideshow.py** code for the TinyMuseumTour Learn guide on HalloWing M4 running Circuit Python 5 Alpha 4. REPL throws error:

`code.py output:
Traceback (most recent call last):
  File "code.py", line 7, in <module>
  File "adafruit_slideshow.py", line 200, in __init__
  File "adafruit_slideshow.py", line 323, in advance
AttributeError: 'Display' object has no attribute 'wait_for_frame'`

Learn guide:
https://learn.adafruit.com/tiny-museum-tour-device/code-with-circuitpython

github code:
https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/master/Slideshows_Soundtrack/slideshows_soundtrack.py

ladyada, 2019-09-19:
@makermelissa see https://github.com/adafruit/Adafruit_CircuitPython_Slideshow/commits/master

makermelissa, 2019-09-19:
Odd, my error is:
```
Traceback (most recent call last):
  File "code.py", line 10, in <module>
AttributeError: 'module' object has no attribute 'TOUCH1'
```
I did a dir(board) and got:
```
>>> import board
>>> dir(board)
['__class__', 'A0', 'A1', 'A2', 'A3', 'A4', 'A5', 'A6', 'A7', 'A8', 'A9', 'ACCELEROMETER_INTERRUPT', 'BATTERY', 'CAP_PIN', 'D0', 'D1', 'D10', 'D11', 'D12', 'D13', 'D2', 'D3', 'D4', 'D5', 'D6', 'D7', 'D8', 'D9', 'DISPLAY', 'EXTERNAL_NEOPIXEL', 'I2C', 'LIGHT', 'MISO', 'MOSI', 'NEOPIXEL', 'RX', 'SCK', 'SCL', 'SDA', 'SENSE', 'SPEAKER_ENABLE', 'SPI', 'TX', 'UART', 'VOLTAGE_MONITOR']
```
So it's not seeing the touch pads for some reason. This was with CP 5.0 Alpha 4 that I downloaded off of circuitpython.org.

makermelissa, 2019-09-19:
Also, I couldn't find any calls to `wait_for_frame()` in the latest Adafruit_CircuitPython_Slideshow.

makermelissa, 2019-09-19:
Some more info. It works fine on HalloWing M0. On HallWing M4, it shows the first image briefly before giving the TOUCH1 error and it's small and in the upper left corner (resolution difference).

makermelissa, 2019-09-19:
Ok, I changed board.TOUCH1 to board.A5 and board.TOUCH4 to board.A2 as labeled on the board. Now this is the error I get:
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 10, in <module>
ValueError: No pulldown on pin; 1Mohm recommended
```

closed 2019-09-19 by ladyada

ladyada, 2019-09-20:
@djecken plz try!

djecken, 2019-09-20:
confirmed works on HalloWing M4 running circuit python 5 alpha 4

PR #848 (2019-09-19): Updated Slideshow_Soundtrack to run on both HalloWing M0 and M4
Please note that TOUCH1-TOUCH4 and SPEAKER alias pins were recently pushed to the HALLOWING M4 (today) and until the next release of CircuitPython may need to be adjusted manually. I tested with a speaker on both the HalloWing M0 and M4. Also, images are smaller on the M4 due to the higher resolution screen.

Fixes #846.