adafruit/Adafruit_Learning_System_Guides #847: HalloWing Jump Sound Errors

djecken, 2019-09-19:
Testing code with HalloWing M4 running Circuit Python 5 alpha 4. 

Link to code:
https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/master/Hallowing_Jump_Sound

Link to guide:
https://learn.adafruit.com/last-minute-halloween-accoutrements-with-hallowing/sound-costume

REPL throws following error:

`code.py output:
Traceback (most recent call last):
  File "code.py", line 43, in <module>
ValueError: No pulldown on pin; 1Mohm recommended`

ladyada, 2019-09-19:
pleaset test it on a hallowing m0 to verify it still works there

djecken, 2019-09-19:
tested code with HalloWing M0 running circuit python 5 alpha 4. Error from REPL

`code.py output:
Traceback (most recent call last):
  File "code.py", line 43, in <module>
ValueError: TFT_BACKLIGHT in use`

djecken, 2019-09-19:
getting very similar error on circuit python 4.1.0 stable 

`Traceback (most recent call last):
  File "code.py", line 43, in <module>
ValueError: TFT_BACKLIGHT in use`

ladyada, 2019-09-19:
@makermelissa will check this out next!

makermelissa, 2019-09-19:
Got the TFT backlight working, but the plot thickens. We have a LIS3DH on the M0 and an MSA301 on the M4 according to the schematic.

PR #851 (2019-09-19): Fixed HalloWing_Jump_Sound example to work on M4 and M0
Tested on HalloWing M4 with CP 5 Alpha 4, HalloWing M0 with CP 5 Alpha 4 and CP 4.1.

Fixes #847.
closed 2019-09-19 by ladyada