adafruit/Adafruit_Learning_System_Guides #876: `pixel.setPixelColor` interfering with Serial communication on Teensy

embedded-creations, 2019-10-08:
I'm trying this sketch with several Teensys: LC, 4.0, 3.5, and found it couldn't even forward the message the ESP32 prints out after resetting to boot mode without dropping characters.  Attempting programming with `ESPTool.py` would fail as it couldn't even get past trying to connect.

If I comment out the `pixel.setPixelColor()` calls in `loop()`, it works without dropping characters, and programming flash (at least on 3.5, I haven't tried the other Teensys yet).

Arduino 1.8.9
Teensyduino 1.4.7
Adafruit_Neopixel 1.2.5

ladyada, 2019-10-08:
huh weird! ok please submit a PR, you can move the LED blinkin' outside the loop

ladyada, 2021-05-01:
closing as no PR was submitted

closed 2021-05-01 by ladyada