adafruit/Adafruit_Learning_System_Guides #879: M4_eyes voice not working on Monster Mask

jerryneedell, 2019-10-11:
I just tried M4_eyes on Monster mask from the updated Learning Guide and "voice" seems to be broken. If I enable it, I just get blank screens, nothing..  Without voice, everything is working fine. Is this a known issue?
I tried this example: https://learn.adafruit.com/adafruit-monster-m4sk-eyes/voice-changer-2#example-config-dot-eye-file-14-8 
as well as just adding voice : true  to other working "eyes"
In both cases, it just seems to hang during the start-up

ladyada, 2019-10-11:
what if ya try the UF2?

jerryneedell, 2019-10-11:
which .uf2?  the default does not have voice enabled.

jerryneedell, 2019-10-11:
ahh -- never mind -- that is a coffin parameter -- will try it

jerryneedell, 2019-10-11:
Is the UF2 from the latest repo -- it used to work

ladyada, 2019-10-11:
https://learn.adafruit.com/adafruit-monster-m4sk-eyes/m-eyes-firmware this Uf2

jerryneedell, 2019-10-11:
yes -- it works -- was it built from the latest repo ?

ladyada, 2019-10-11:
no, repo is undergoing some work - could be something missing.

jerryneedell, 2019-10-11:
OK -- thanks -- I'll wait for updates and take a closer look at the code.

jerryneedell, 2019-10-22:
it is working with the latest code ( with Adafruit_ZeroPDM) - tested on 10/220/2019

closed 2019-10-22 by jerryneedell