adafruit/Adafruit_Learning_System_Guides #890: Dead link in Weather Pixes

joeynuggetz, 2019-10-20:
Code links to query.yahooapis.com which doesn't exist anymore.

ladyada, 2019-10-20:
please be specific where exactly you see a broken link - thanx

joeynuggetz, 2019-10-20:
Line 17 

https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/master/ESP8266_Weather_Pixels/WeatherPixels/weatherPixels.ino

ladyada, 2019-10-20:
ok sounds like the code is not going to work anymore 
@TheKitty please deprecate the guide https://learn.adafruit.com/feather-weather-lamp

joeynuggetz, 2019-10-20:
Darn, looked cool.

ladyada, 2019-10-20:
if you find an alternative solution to the code please submit a PR, thanks!

TheKitty, 2019-10-23:
@ladyada - the guide has now been marked as deprecated and a red warning placed that the code no longer works due to the yahoo change.

closed 2019-10-23 by ladyada