adafruit/Adafruit_Learning_System_Guides #891: PyPortal_Smart_Thermometer fails in displayio with CP 5.0-alpha

jerryneedell, 2019-10-21:
This fails for CP5.0-alpha.  It works for CP 4.1.0

```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-alpha.4-167-g0ccab508f on 2019-10-11; Adafruit PyPortal with samd51j20
>>> import thermometer
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "thermometer.py", line 59, in <module>
  File "thermometer_helper.py", line 33, in __init__
ValueError: Group already used
>>> 

```

ladyada, 2019-10-21:
jerry, can you try narrowing the error to an example that we can easily repro for analysis? it would be super helpful!

jerryneedell, 2019-10-21:
Sure -- I'm trying to do that, not as easy as I'd hoped. I'll post it when I have something.

jerryneedell, 2019-10-21:
Yay! fixed -- PR soon after  bit more testing. Thanks @tannewt

PR #894 (2019-10-22): revise PyPortal_Smart_Thermometer thermometer_helper.py 
fixes #891 
remove unnecessary appends and calls to refresh_soon and wait_for_frame

tested on CP4.10. and CP5.0alpha-4
closed 2019-10-22 by brentru