adafruit/Adafruit_Learning_System_Guides #903: List of required Arduino libraries in Adafruit MONSTER M4SK guide is not sufficient.

vatbrain, 2019-10-25:
The pdf overview guide for the Adafruit MONSTER M4SK dated 2019-10-21, as well as the corresponding website lists the following libraries as being required to build M4_Eyes:
 Adafruit_GFX, Adafruit_ST7789, Adafruit_ZeroDMA
 Adafruit_ZeroPDM, Adafruit_ImageReader, Adafruit_SPIFlash
 Adafruit_TinyUSB, SdFat - Adafruit fork (not the standard SdFat fork)
 ArduinoJson (not Arduino_JSON)

The above list is not sufficient. The following libraries are also required in order to build:
 Adafruit_AM2320_sensor_library     
 Adafruit_Arcada_Library             
 Adafruit_TouchScreen
 Adafruit_Unified_Sensor
 Adafruit_ImageReader_Library        
 Adafruit_LIS3DH                     
 Adafruit_NeoPixel
 Adafruit_ZeroTimer_Library
 Adafruit_seesaw_Library             

In addition, it was necessary for me to pull the latest Adafruit_ZeroPDM library from GitHub, as the version through the Arduino Library Manager was missing some recent updates.

It would be nice to have the documentation updated to save people time in the future.

ladyada, 2019-10-25:
oh yah we did a refactor lately, so more libraries are needed
@PaintYourDragon wanna add those to the guide? ill check the zeropdm release

PaintYourDragon, 2019-11-02:
I don’t see anywhere that AM2320 is required, but the rest have been added to the guide, thanks.

closed 2019-11-02 by PaintYourDragon