adafruit/Adafruit_Learning_System_Guides #910: Error with Circuit Python Flying Toaster

djecken, 2019-10-30:
Getting error with flying toaster code. Itsy Bisy M4 running Adafruit CircuitPython 5.0.0-alpha.4 on 2019-09-15; Adafruit ItsyBitsy M4 Express with samd51g19

Using latest library from circuit python bundle adafruit-circuitpython-bundle-5.x-mpy-20191024

`code.py output:
Traceback (most recent call last):
  File "code.py", line 177, in <module>
  File "code.py", line 120, in advance_animation
KeyboardInterrupt: `

@dastels

ladyada, 2019-10-30:
keyboard interrupt means you pressed control-C or soemthing, please try again?

djecken, 2019-10-30:
doesn't seem to start. different error message on display 

[image]

ladyada, 2019-10-30:
ooh ok @makermelissa wanna take a look at this - you know how to fix? (@jerryneedell did a few of these but its good for you to do as well :)

djecken, 2019-10-30:
getting very similar error with Scrolling Clouds code as well.

`AttributeError: 'ST7789' object has no attribute 'refresh_soon'`

https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/master/CircuitPython_Scrolling_Clouds

makermelissa, 2019-10-30:
I think just removing the line that calls that function is what's needed, but sure I can.

dastels, 2019-10-30:
That was written a few months ago... the display system has likely 
changed a bit since.

makermelissa, 2019-10-30:
Which guide is this for? I could only find an Arduino Scrolling Clouds guide.

djecken, 2019-10-30:
@makermelissa no guide yet, flying toaster code is here https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/master/CircuitPython_Flying_Toasters

makermelissa, 2019-10-30:
Oh, ok. Thanks. I'll infer the wiring from the code then.

ladyada, 2019-11-01:
is this resolved?

makermelissa, 2019-11-01:
Oh yeah, I forgot to close it.

closed 2019-11-01 by makermelissa