adafruit/Adafruit_Learning_System_Guides #913: TFT Gizmo Snow Globe

djecken, 2019-11-01:
Testing out **snowglobe_fancy.py** code with Adafruit CircuitPython 5.0.0-alpha.4 on 2019-09-15; Adafruit Circuit Playground Bluefruit with nRF52840. Using library bundle adafruit-circuitpython-bundle-5.x-mpy-20191024

Error from REPL:

`code.py output:
Traceback (most recent call last):
  File "code.py", line 51, in <module>
ValueError: color buffer must be 3 bytes (RGB) or 4 bytes (RGB + pad byte)`

the **snowglobe_simple.py** code runs fine, no errors.

link to code: 
https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/master/TFT_Gizmo_Snowglobe

djecken, 2019-11-01:
maybe because the bitmaps are missing?

makermelissa, 2019-11-01:
Looks like if the bitmap is missing, it's trying to load the filename into a property that expects a number.

djecken, 2019-11-01:
@makermelissa yep, looks like it. @caternuson hiya, feel free to add those bitmaps when ya can :-)

caternuson, 2019-11-01:
Yep. Just code here. I'll add the bitmaps to the guide.

I tested that behavior and it seemed to be fault tolerant - like it was able to cast the string to an int and at least have some kind of color background. Let me look at that again.... :/

djecken, 2019-11-01:
works nicely!

closed 2019-11-01 by djecken

caternuson, 2019-11-01:
cool. thanks. sorry about that.