adafruit/Adafruit_Learning_System_Guides #970: CircuitPlaygroundExpress_AudioSine.py: loop value doesn't appear to have an effect

Abraxas3d, 2019-12-15:
Greetings all!

CircuitPlaygroundExpress_AudioSine.py seems to run only once regardless of the value of "loop". 

The loop=True doesn't appear to have any effect. 

With loop=True, The sample plays once and then the program halts. 
Changing it to loop=False resulted in the same behavior. The sample played once and the program halted.
Removing loop=Value entirely resulted in the same behavior. The sample played once and the program halted.

kattni, 2019-12-16:
There are a few issues going on here:

1. The use of the word `loop` here refers to looping the sine wave sample. A single sine wave sample is a few hundredths of a second long, and loop in this case is referring to continuing to loop that incredibly short sample for the duration of the `time.sleep()`, not continuously looping it forever.
2. The comment in this code with regards to the looping behavior is unclear - it seems to imply that the sample will loop forever, which is not the case. Part of this is that "loop" has multiple meanings when combining audio and Python programming.
3. There is a bug with the nRF52840 where it treats `loop` as `True` with `RawSample` regardless of what it is set to (see: https://github.com/adafruit/circuitpython/issues/2078). If you were to set loop to False on a SAMD chip (for example, Circuit Playground Express), you would hear a very quick burst of sound, and silence for the rest of the `time.sleep(1.0)`. On the nRF, it plays for the duration of the `sleep` regardless of being `True` or `False`. 

I will update this code to have a more clear explanation of what `loop=True` is doing.

Until the bug is fixed, if you want to have a specific length for the tone, change the duration of the `time.sleep()`. If you want it to play continuously, use a `while` loop and do not call `audio.stop()`. 

As the bug is with CIrcuitPython, and not this program, I will close this issue once the comment is updated.

Abraxas3d, 2019-12-16:
Thank you @kattni, super helpful! Looking to show off the Circuit Playground Bluefruit at a hardware meetup tonight.

kattni, 2019-12-16:
Comment has been updated to clarify what is meant by looping. We are currently looking into the audio bug present on the nRF boards.

@Abraxas3d Thanks for letting us know!

closed 2019-12-16 by kattni