adafruit/Adafruit_Learning_System_Guides #1006: Not immediately obvious that "Hallowing" examples won't work on the Hallowing M4

jin-eld, 2020-01-28:
Hi,

I had a bit of a rough start trying to get going on a Hallowing M4, wanted to start with something really simple, so I went through the "Hallowing" directories in the repo. Hallowing/graphicstest compiled, but did not show anything at all, not even serial printouts which left me wondering... Went on trying the other examples, most of them did not even compile.

Only then I got a hint on IRC that they are probably not meant to work on the M4 at all and this indeed seems to be the case, and I was able to build and run M4_Eyes which proved the above point.

I don't know if I missed that, but would be cool to have a more prominent warning somewhere that "Hallowing" examples won't work on the "Hallowing M4".

ladyada, 2020-02-02:
hi - yep the best guide for hallowing m4 is https://learn.adafruit.com/adafruit-hallowing-m4
if you could submit some readme PR's to the directories you think are confusing (without renaming any existing dirs/files as that will break embedding) that would be awesome!

jin-eld, 2020-02-02:
> hi - yep the best guide for hallowing m4 is https://learn.adafruit.com/adafruit-hallowing-m4
it was indeed helpful, but once I had the IDE set up and running according to the guide I looked at the Adafruit_Learning_System_Guides repo, because I wanted to try out more examples.

I noticed these directories:
Hallowing
Hallowing_Case
HalloWing_Cat_Toy
Hallowing_Googly_Eye
HalloWing_Jump_Scare_Trap
Hallowing_Jump_Sound
Hallowing_Lightsaber
Hallowing_Mac
Hallowing_Minotaur_Maze
Hallowing_Spirit_Board
HalloWing_Tour_Guide

And I did not realize the above were not for the M4. 

> if you could submit some readme PR's to the directories you think are confusing (without renaming any existing dirs/files as that will break embedding) that would be awesome!

I can do that, but I need a hint first - which Hallowing are the above examples supposed to be running on?

I'd add something like "NOTE: this example is for the Hallowing xyz and not for the Hallowing M4", just need to be sure I know what to put in those readmes :) Are those for the M0 or was there another Hallowing before that?

ladyada, 2020-02-03:
probably none of them work - there's only hallowing m0 and hallowing m4

jin-eld, 2020-02-03:
Oh... so what would you like me to write in the readme then? "This code does not work" would be kind of odd.. would id be better to create a "staging" directory and move the non working examples there?

ladyada, 2020-02-03:
hi, its whatever you want to write - this is your issue :) don't move any files around, you can only create a readme if you like!

makermelissa, 2020-02-03:
Hallowing_Jump_Sound was updated to work with both the HalloWing M0 and M4. The others, I think haven't been updated.

jin-eld, 2020-02-03:
> hi, its whatever you want to write - this is your issue :) don't move any files around, you can only create a readme if you like!

doh... we're talking past each other a bit :) I don't know what this code is intended for, if you were planning to fix it or what it is all about or why the directories are called "Hallowing", but the examples do not actually work. Sure I can submit something like "I have no clue what it is, beware that it does not work!", I was hoping to come up with something more informative - but I don't have the background knowledge there.

> Hallowing_Jump_Sound was updated to work with both the HalloWing M0 and M4. The others, I think haven't been updated.

"Updated" means these examples have been written for some other boards, right? That's what I was actually asking: for which boards? Then I could put that in the readme, i.e. "this eaxmple will work with board xyz, but not with Hallowing M0/M4", this was the point of my original question :)

ladyada, 2020-02-03:
jin, these examples are all used in guides - they are embedded from github into the learning system. they're not really intended to be browsed because there's no wiring diagrams included, they're backend storage. if you'd like to add readme's we're happy to look at a PR - most folks to dont look in the folders.

jin-eld, 2020-02-03:
Oooh now I get the picture, thanks for clearing this up, now I understand why it was important not to change the directory structure. So this repo was not intended for browsing at all and there was no guarantee for any of them to work unless they were linked to the learning pages. Got it.

ladyada, 2021-05-01:
closing as no PR was submitted

closed 2021-05-01 by ladyada