adafruit/Adafruit_Learning_System_Guides #1010: PropMaker_Lightsaber audio issues

AmpCat, 2020-01-30:
I just downloaded and used the build uploaded on 28Jan20, and while it's improved the idle audio playback dramatically, there's still a glitch in the idle audio every once in a while. It ranges from a quick pop to an oscillating sort of white noise. Not sure what the root of this is. Other people have also reported it. Not sure if this is a software playback issue or electrical problem, but it's only in the idle sound as far as I can tell. There was a comment about audio file length potentially being an issue.

jepler, 2020-01-31:
Hi, I just want to let you know that I've reproduced a problem similar to what you describe, though I used a modified version of the script on a Metro M4 Express with a PAM8302 breakout board, rather than the guide's suggested hardware.  I hear very small pops from time to time, and then once every 2 minutes or so there is a stretch of completely garbled sound.  I think that the bug lies in the core of CircuitPython, though a modified wave file may work around or at least minimize the bug's effects.

If you can, please let me know whether this wave file improves things.  This would be helpful in confirming or refuting my theory:

[idle-trimmed.zip](https://github.com/adafruit/Adafruit_Learning_System_Guides/files/4137414/idle-trimmed.zip)

AmpCat, 2020-01-31:
Okay, I'll give it a shot this evening.

I was also able to get some improvements using audioio.Mixer to loop the
idle sound and mix the other sounds on top of it. In this situation all the
small pops and occasional garbles go away. I can send the example when I
get home if you'd like to try.

jepler, 2020-02-27:
Hi, we believe that #1037 fixes this.  For users newly visiting the guide, they will be linked to updated wave files.

closed 2020-02-27 by jepler