adafruit/Adafruit_Learning_System_Guides #1028: CLUE Cycling Simple – Type Errorr: num_lines

djecken, 2020-02-11:
Testing code from learn guide: https://learn.adafruit.com/bluetooth-bicycle-speed-cadence-sensor-display-with-clue/code-the-clue-in-circuitpython

Running Adafruit CircuitPython 5.0.0-beta.5 on 2020-02-05; Adafruit CLUE nRF52840 Express with nRF52840

Using libraries from latest bundle beta 5, 20200211

`code.py output:
Traceback (most recent call last):
  File "code.py", line 14, in <module>
TypeError: unexpected keyword argument 'num_lines'`

jedgarpark, 2020-02-17:
@kattni do you know if something changed in the adafruit_clue library? The error is from this line:
`clue_data = clue.simple_text_display(title="Cycle Revs", title_scale=1, text_scale=3, num_lines=3)
`
If I remove the `num_lines=3` argument this works as expected.

caternuson, 2020-02-17:
Looks like there were some breaking changes done around 2.0.0. See first bullet:
https://github.com/adafruit/Adafruit_CircuitPython_CLUE/releases/tag/2.0.0

kattni, 2020-02-18:
@jedgarpark What Carter said. The library now automatically creates the necessary number of lines based on the code. `num_lines` is no longer necessary or allowed.

jedgarpark, 2020-02-18:
Thanks, I'll update this project code.

jedgarpark, 2020-02-18:
the code in learn guide repo is now fixed, thanks for the catch @djecken !

closed 2020-02-18 by jedgarpark