adafruit/Adafruit_Learning_System_Guides #1091: bug with LONNNNNNNG quotes

TonyLHansen, 2020-04-27:
The code for the FeatherWing Quote Display has a bug that shows up when using a Huzzah32. If the quote happens to more than 261 characters, you get a board fault. The test string on line 195 is about 40 characters too short to demonstrate the issue. The current set of quotes happens to have two that are longer than that test string.

ladyada, 2020-04-27:
hi please indicate the exact project URL

TonyLHansen, 2020-04-27:
my apologies. The problem is in Feather_ePaper_Quotes, found in [https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/master/Feather_ePaper_Quotes](Feather_ePaper_Quotes) and as described in [https://learn.adafruit.com/epaper-display-featherwing-quote-display](epaper-display-featherwing-quote-display).

The value for the longest string appears to be less deterministic than I though. I just downloaded a fresh copy of adafruit_feather_quote, uncommented the jsonstring value on line 195, and started appending characters to the text field within it, as in "--23456789--23456789--23456789--23". At some point it will start generating faults such as  this:

`
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core 1 register dump:
PC      : 0x401600c1  PS      : 0x00060636  A0      : 0x801601b8  A1      : 0x3ffb0070  
A2      : 0x00000000  A3      : 0x00000164  A4      : 0x00000164  A5      : 0x00000000  
A6      : 0x00000164  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffc1224  
A10     : 0x3ffc1224  A11     : 0x00005118  A12     : 0x00000027  A13     : 0x00000164  
A14     : 0x00000164  A15     : 0x40160164  SAR     : 0x00000010  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x401600c1:0x3ffb0070 0x401601b5:0x3ffb00b0 0x401601d6:0x3ffb00d0 0x401601d6:0x3ffb00f0 0x401601d6:0x3ffb0110 0x401601d6:0x3ffb0130 0x401601d6:0x3ffb0150 0x401601d6:0x3ffb0170 0x401601d6:0x3ffb0190 0x401601d6:0x3ffb01b0 0x401601d6:0x3ffb01d0 0x401601d6:0x3ffb01f0 0x401601d6:0x3ffb0210 0x401601d6:0x3ffb0230 0x401601d6:0x3ffb0250 0x401601d6:0x3ffb0270 0x401601d6:0x3ffb0290 0x401601d6:0x3ffb02b0 0x401601d6:0x3ffb02d0 0x401601d6:0x3ffb02f0 0x401601d6:0x3ffb0310 0x401601d6:0x3ffb0330 0x401601d6:0x3ffb0350 0x401601d6:0x3ffb0370 0x401601d6:0x3ffb0390 0x401601d6:0x3ffb03b0 0x401601d6:0x3ffb03d0 0x401601d6:0x3ffb03f0 0x401601d6:0x3ffb0410 0x401601d6:0x3ffb0430 0x401601d6:0x3ffb0450 0x401601d6:0x3ffb0470 0x401601d6:0x3ffb0490 0x401601d6:0x3ffb04b0 0x401601d6:0x3ffb04d0 0x401601d6:0x3ffb04f0 0x401601d6:0x3ffb0510 0x401601d6:0x3ffb0530 0x401601d6:0x3ffb0550 0x401601d6:0x3ffb0570 0x401601d6:0x3ffb0590 0x401601d6:0x3ffb05b0 0x401601d6:0x3ffb05d0 0x401601d6:0x3ffb05f0 0x401601d6:0x3ffb0610 0x401601d6:0x3ffb0630 0x401601d6:0x3ffb0650 0x401601d6:0x3ffb0670 0x401601d6:0x3ffb0690 0x401601d6:0x3ffb06b0 0x401601d6:0x3ffb06d0
`

This particular stack trace points at
0x401601d6: Adafruit_GFX::drawFastHLine(short, short, short, unsigned short) at /Users/tonyhansen/Documents/Arduino/libraries/Adafruit_GFX_Library/Adafruit_GFX.cpp:1960

but I've seen other stack traces too that point elsewhere.

ladyada, 2020-04-27:
@cogliano please check it out

cogliano, 2020-04-27:
Thanks, I'll take a look.

cogliano, 2020-04-27:
In line #93, I increased the buffer size to 1024 and that fixed the issue. I have submitted the change but you can make the change now if you don't want to wait to have it posted.

TonyLHansen, 2020-04-27:
Aha, I missed seeing that static buffer declaration. Having a small size for buff would definitely cause issues.

I suggest you also the strcpy() statement a few lines later from this:

`
      strcpy(buff,str);
`
to this:
`
   if (strlen(str) >= sizeof(buff)-1) {  // protect against buffer overflow
      strncpy(buff, str, sizeof(buff)-2);
      buff[sizeof(buff)-1] = '\0';
    } else {
      strcpy(buff,str);
    }
`
That will also future proof the code.

Because of the small memory on the microcontroller, you could then drop the size of buff back down a bit so you don't have such a large buffer floating around that is empty most of the time.

ladyada, 2020-05-18:
@TonyLHansen @cogliano closing as this seems resolved

closed 2020-05-18 by ladyada

TonyLHansen, 2020-05-25:
This was not dealt with.

I suggest you also the strcpy() statement a few lines later from this:

      strcpy(buff,str);

to this:

   if (strlen(str) >= sizeof(buff)-1) {  // protect against buffer overflow
      strncpy(buff, str, sizeof(buff)-2);
      buff[sizeof(buff)-1] = '\0';
    } else {
      strcpy(buff,str);
    }

That will also future proof the code.

Because of the small memory on the microcontroller, you could then drop the size of buff back down a bit so you don't have such a large buffer floating around that is empty most of the time.