adafruit/Adafruit_Learning_System_Guides #1110: PyPortal_MQTT_Control: not working on CircuitPython 5.3

mikewop, 2020-05-18:
It seems the library MiniMQTT has changed. The code in PyPortal_MQTT_Control uses 

`# Set up a MiniMQTT Client
client = MQTT(socket,
              broker = secrets['broker'],
              port = 1883,
              username = secrets['user'],
              password = secrets['pass'],
              network_manager = wifi)`
But the latest MiniMQTT library from adafruit-circuitpython-bundle-py-20200516.zip 

has a signature of:
`    # pylint: disable=too-many-arguments,too-many-instance-attributes, not-callable, invalid-name, no-member
    def __init__(
        self,
        broker,
        port=None,
        username=None,
        password=None,
        client_id=None,
        is_ssl=True,
        log=False,
        keep_alive=60,
    ):`

Notice that "socket" and "network_manager" are no longer part of the constructor.

Error:
`Traceback (most recent call last):
  File "code.py", line 205, in <module>
TypeError: function got multiple values for argument 'broker'
`
The code seems to be out of date w.r.t to the latest version of minimqtt

brentru, 2020-05-18:
@mikewop   Thanks for reporting this issue., a similar issue is open on MiniMQTT (https://github.com/adafruit/Adafruit_CircuitPython_MiniMQTT/issues/25) and I will be taking a look at it later today.

brentru, 2020-05-19:
@mikewop I patched MiniMQTT's hostname (https://github.com/adafruit/Adafruit_CircuitPython_MiniMQTT/pull/33). However, this guide's code is still not up to date w.r.t the most recent version of MiniMQTT.

@ladyada Should I ping Richard A. (https://learn.adafruit.com/pyportal-mqtt-sensor-node-control-pad-home-assistant) who authored this guide? I could update the code but would not be able to test it as I don't have Home Assistant set up.

ladyada, 2020-05-20:
@brentru yep please contact 'em to update the guide, if you can submit the PR then they could test it!