adafruit/Adafruit_Learning_System_Guides #1127: revisiting #1091 on Feather_ePaper_Quotes

TonyLHansen, 2020-05-25:
The "reopen button" is not available. Regarding #1091, which is about Feather_ePaper_Quotes, found in https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/master/Feather_ePaper_Quotes and as described in https://learn.adafruit.com/epaper-display-featherwing-quote-display. The "fix" that was merged previously for this changed one fixed-length buffer for a slightly larger one. Instead of fixing the problem, it just postpones it until the new buffer size overflows.

I suggest you also change the strcpy() statement a few lines after the declaration on line 93

  static char buff[1024];
. . .
  strcpy(buff,str);

to this:

  static char buff[512];
. . .
if (strlen(str) >= sizeof(buff)-1) { // protect against buffer overflow
strncpy(buff, str, sizeof(buff)-2);
buff[sizeof(buff)-1] = '\0';
} else {
strcpy(buff,str);
}

That will future proof the code.

Because of the small memory on the microcontroller, I also drop the size of buff back down a bit so you don't have such a large buffer floating around that is empty most of the time.

I'll put together a PR with this fix.

ladyada, 2020-05-25:
@cogliano please look!

TonyLHansen, 2020-05-25:
Well, github isn't letting me create another fork of Adafruit_Learning_System_Guides into my home tree and I don't have permissions to create a PR directly into adafruit/Adafruit_Learning_System_Guides. (I have a pending PR from last November that hasn't been dealt with yet, for issue #915.) So I can't create a PR for this specific issue. Sorry.