adafruit/Adafruit_Learning_System_Guides #1209: Circuit Python projects that use adafruit_button library should be updated to use new API

FoamyGuy, 2020-08-16:
In the adafruit_button library PR [#23](https://github.com/adafruit/Adafruit_CircuitPython_Display_Button/pull/23) the button API was changed slightly so that the button itself can be added to Groups and Displays rather than needing to add the `group` property like:  `button.group`. 

The code is backward compatible, and the old version prints a deprecation warning.

I put together a little script to search the projects in the repo to find ones that make use of adafruit_button 
```python
import os
for file in os.listdir("."):
    #print(file)
    try:
        for sub_file in os.listdir(file):
            #print(f"\t{sub_file}")
            if sub_file.endswith(".py"):
                script_file = open(f"{file}/{sub_file}", "r")
                script_text = script_file.read()
                script_file.close()

                if "adafruit_button" in script_text:
                    print(f"{file}/{sub_file}")
    except NotADirectoryError:
        pass
```
here are the ones I found:
```
PyGamer_NeoPixel_Strip_Control/code.py
PyPortal_Calculator/code.py
PyPortal_EZ_Make_Oven/code.py
PyPortal_Guitar_Tuner/code.py
PyPortal_LIFX_Controller/code.py
PyPortal_MQTT_Control/code.py
PyPortal_NeoPixel_Color_Picker/PyPortal_NeoPixel_Color_Picker.py
PyPortal_Philips_Hue_Controller/code.py
PyPortal_Titano_Weather_Station/code.py
PyPortal_TOTP_Friend/code.py
PyPortal_User_Interface/code.py
```

It's not quite urgent since the backward compatibility is provided so these will not break. I can work my way through this list. 

But it may also serve as a "good first issue for someone" so I'll tag it. If any finds this and wants to work on it and needs help feel free to reach out to me here or on discord.

nezra, 2021-01-19:
This actually is broken now on the pyportal titano with circuitpython 6.0. even though the deprecation message is provided, the images itself do not display at all from the user_interface example here:

https://learn.adafruit.com/making-a-pyportal-user-interface-displayio/the-full-code

FoamyGuy, 2021-01-20:
I ran the code in that guide with:
```
Adafruit CircuitPython 6.1.0-rc.1 on 2021-01-15; Adafruit PyPortal with samd51j20
```
and all of the buttons were shown properly for me. 

Which one(s) are not being shown for you?

nezra, 2021-01-20:
No button, no background. I have the pyportal titano if that makes a difference. I'm on 6.0.1 stable

nezra, 2021-01-25:
`Adafruit CircuitPython 6.0.1 on 2020-12-28; Adafruit PyPortal Titano with samd51j20`

Nevermind! found the issue. 

The default code sets the brightness to "0.3". On the Titano, whenever the brightness is set to less than 0.5, it turns the display off completely in that sample code. I'm not sure where to report that since it appears to be an issue with board.DISPLAY.brightness. i tested it outside of the learning code by itself, and the same issue occurs.

FoamyGuy, 2021-01-25:
Perhaps the Titano treats brightness differently than the others, I'm not sure. I think the main circuitpython repo would be the best spot for the larger potential issue with brightness. 

But knowing that it is occuring it might also be good to change the sample code so that it will work as close to expected as possible on all of variations of the device.

I did make a PR here that updates the bottom API code in this example if you're interested in trying it out. (Don't think it will have any affect on the brightness situation though) https://github.com/adafruit/Adafruit_Learning_System_Guides/pull/1422