adafruit/Adafruit_Learning_System_Guides #1281: Why does bme280_data_bytes exist?

loganwedwards, 2020-10-22:
https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/ec740ac4eccab0c67f651e9ecdf23148a33d46e6/lorawan_sensing_network/lora_device.py#L79

I think the docs can be clearer here. Does this variable exist solely for debugging or is it meant to be the actual payload.

Currently `rfm9x.send(bme280_data)` is listed.

To be clear, I am happy to make a PR for this, but wanted to fully understand the intended usage.

ladyada, 2020-10-22:
probably a typo @brentru can look

brentru, 2020-10-22:
@loganwedwards  Yes, that's a typo - it's meant to be the actual payload in L83. The RFM9x.send() function operates on bytes (https://github.com/adafruit/Adafruit_CircuitPython_RFM9x/blob/master/examples/rfm9x_simpletest.py)

> To be clear, I am happy to make a PR for this, but wanted to fully understand the intended usage.
That'd be great, please tag me.

loganwedwards, 2020-10-22:
Great. Thanks @ladyada and @brentru !

PR #1283 (2020-10-23): Update payload variable in send function
**What Changed**
This change simply changes the payload in `rfm9x.send()` to the expected argument, `bme280_data_bytes`. @brentru 
This fixes #1281 
**Tests**
No matching test suites were found, but I did look at the source code and noticed that it is, in fact, expecting some bytes. See [here](https://github.com/adafruit/Adafruit_CircuitPython_RFM9x/blob/6590c91a9146f41a7984a52febce9b21d0df8748/adafruit_rfm9x.py#L599)
closed 2020-10-23 by brentru