adafruit/Adafruit_Learning_System_Guides #1340: Issue compiling the "teensySNES_test1.ino"

cheikhd, 2020-12-03:
Trying to compile the "teensySNES_test1.ino" to see how it works, , i've got the error   : 
 " 'KEY_U' was not declared in this scope", think that is missing some key.h file to compile ?!

ladyada, 2020-12-07:
@cheikhd ok fixed in a PR, we didnt actually try the code (its from like 2010 or something) but it at least compiles now

@paulstoffregen is there anyone who'se gotten github Actions to install teensy over arduino-cli so we can have our teensy code auto-checked for compilation errors & warnings? :)

closed 2020-12-07 by ladyada

PaulStoffregen, 2020-12-08:
@ladyada - Oh, yeah, a couple years ago I added a "headless" option to the Teensyduino installer, which was meant for your CI script.  Then of course this completely fell off my radar while bringing up new hardware.  Back then, at least as I recall, your CI script installed a copy of Arduino IDE 1.8.5 into virtual machine, and ran it from command line.  My installer is meant to work with that sort of usage.  It needs a copy of the Arduino IDE to install into.  All the files go into the IDE's "hardware" folder.

If you're now using arduino-cli, I'm afraid Teensy isn't currently supporting the non-IDE command line.  The main issue is arduino-cli looks only in your ~/.arduino15 folder for all installed platforms.  It doesn't look in the "hardware" folder of any particular copy of the IDE.  Or at least not as far as I know.  Maybe it can somehow?

Or maybe a script could be made to temporarily extract a copy of the IDE, download the Teensyduino installer and have it put all the files into that IDE, and then copy stuff from the IDE's hardware folder over to the ~/.arduino15/packages folder?

FWIW, here's the commands which would have gone into a CI script of the type I saw the last time I looked (using the IDE, not arduino-cli).

```
wget https://downloads.arduino.cc/arduino-1.8.13-linux64.tar.xz
wget https://www.pjrc.com/teensy/td_153/TeensyduinoInstall.linux64
wget https://www.pjrc.com/teensy/49-teensy.rules
sudo cp 49-teensy.rules /etc/udev/rules.d/
tar -xf arduino-1.8.13-linux64.tar.xz
chmod 755 TeensyduinoInstall.linux64
./TeensyduinoInstall.linux64 --dir=arduino-1.8.13
```

If you can point me to the CI script you're using now, I'll take another look....

ladyada, 2020-12-08:
Yeah we moved away from installing a headless arduino IDE because it took a very long time and was tough to maintain. 
We install the arduino-cli here
https://github.com/adafruit/ci-arduino/blob/master/actions_install.sh
and then we install the actual BSP here
https://github.com/adafruit/ci-arduino/blob/master/build_platform.py#L114

@cmaglie do you have any recommendations on how we could install teensyduino onto arduino-cli?

ubidefeo, 2020-12-08:
@PaulStoffregen 
>arduino-cli looks only in your ~/.arduino15 folder for all installed platforms. It doesn't look in the "hardware" folder of any particular copy of the IDE. Or at least not as far as I know. Maybe it can somehow?

Correct, it looks into the CLI data folder and in `hardware` inside the sketchbook.
Of course where there's a will there's a way 😬

PaulStoffregen, 2020-12-09:
Will work towards a solution for Adafruit CI testing.  I'm a little embarrassed to admit some of Teensy's install is leftover ways from the very early versions of Arduino, before even 1.0.

As I discussed with Cristian and Martion before, I'm reluctant to offer a Boards Manager package for end users until the Arduino IDE supports a Pluggable Serial Monitor feature.  The Java-based IDE serial monitor is also much too slow for how fast Teensy 4.0 can transmit/  You're probably aware of my FifoDocument patches.  Those need to get merged before I can offer Teensy support to users without a special installer which adds that patch.

But merging patches to the IDE is a very slow process.  Or at least it has been historically.  In the meantime, I'll try to come up with a way for my installer to provide what Adafruit's CI script needs.

ladyada, 2020-12-09:
thanks! i think it would be helpful for anyone else who writes teensy-designed code as well, :)