adafruit/Adafruit_Learning_System_Guides #1342: audioio.WaveFile -> audiocore.WaveFile

tannewt, 2020-12-03:
Some guides still use WaveFile in audioio. In 5 it was moved to audiocore and in 6 it was removed from audio.

So, guides should be switched to using audiocore. It'll work on 5.x and 6.x already.

Here is a grep (a few are correct):

```
ag -Q WaveFile Adafruit_Learning_System_Guides/
Adafruit_Learning_System_Guides/HalloWing_Tour_Guide/code.py
53:        wave = audioio.WaveFile(wave_file)

Adafruit_Learning_System_Guides/CPB_ANCS/code.py
17:from audiocore import WaveFile
49:wave = WaveFile(file)

Adafruit_Learning_System_Guides/Burning_Fire_Wizard_Staff/code.py
92:        wave = audiocore.WaveFile(wave_file)

Adafruit_Learning_System_Guides/Hallowing_Lightsaber/lightsaber_unicorn.py
119:        wave = audioio.WaveFile(wave_file)

Adafruit_Learning_System_Guides/Hallowing_Lightsaber/lightsaber_standard.py
54:        wave = audioio.WaveFile(wave_file)

Adafruit_Learning_System_Guides/PropMaker_KeyBlade/code.py
73:        wave = audioio.WaveFile(wave_file)

Adafruit_Learning_System_Guides/CPX_Sound_Box/code.py
40:    with audioio.WaveFile(wave_file) as wave:

Adafruit_Learning_System_Guides/ABC_Soundboards_for_NeoTrellis/code.py
79:        wave = audioio.WaveFile(f)
102:    wav = audioio.WaveFile(f)
120:            wav = audioio.WaveFile(f)
165:            wav = audioio.WaveFile(f)

Adafruit_Learning_System_Guides/Grand_Central_Soundboard/code.py
37:    wav = audioio.WaveFile(audio_file)

Adafruit_Learning_System_Guides/HAL-9000/code.py
30:        wav = audioio.WaveFile(f)

Adafruit_Learning_System_Guides/NeoTrellis_M4_Grains_of_Sand/code.py
80:wav = audioio.WaveFile(f)

Adafruit_Learning_System_Guides/CLUE_Rock_Paper_Scissors/advanced/rps_audio.py
23:from audiocore import WaveFile
40:    _file_buf = None  # Use for WaveFile objects
56:                fhs[file] = WaveFile(wav_file, self._file_buf)

Adafruit_Learning_System_Guides/Animatronic_Hand/binary_counting.py
25:        wav = audioio.WaveFile(f)

Adafruit_Learning_System_Guides/Christmas_Soundboard/code.py
59:        wave = audioio.WaveFile(f)
84:    wav = audioio.WaveFile(f)
105:            wav = audioio.WaveFile(f)
150:            wav = audioio.WaveFile(f)

Adafruit_Learning_System_Guides/Disc_Step_Sequencer/Disc_Step_Sequencer.py
30:    wav = audioio.WaveFile(f)
57:    sample = audioio.WaveFile(wave_file)

Adafruit_Learning_System_Guides/Introducing_CircuitPlaygroundExpress/CircuitPlaygroundExpress_808_Drum_Machine.py
8:    from audiocore import WaveFile
10:    from audioio import WaveFile
46:    wave = WaveFile(file)

Adafruit_Learning_System_Guides/Introducing_CircuitPlaygroundExpress/CircuitPlaygroundExpress_AudioFiles.py
5:    from audiocore import WaveFile
[... 169 lines cut ...]

Adafruit_Learning_System_Guides/Motion_Gift_Box/code.py
34:    wav = audioio.WaveFile(audio_file)

Adafruit_Learning_System_Guides/Hand_Turkey/code.py
41:        wav = audioio.WaveFile(f)

Adafruit_Learning_System_Guides/Music_Box_with_Crickit/code.py
14:audio = audioio.WaveFile(open(AUDIO_FILENAME, "rb"))

Adafruit_Learning_System_Guides/HalloWing_Jump_Scare_Trap/hallowing_jump_scare_trap.py
40:    wave = audioio.WaveFile(wave_file)

Adafruit_Learning_System_Guides/CP101_StateMachines/brute-force/code.py
230:    wav = audioio.WaveFile(sound_file)

Adafruit_Learning_System_Guides/CP101_StateMachines/classes/code.py
194:    wav = audioio.WaveFile(sound_file)

Adafruit_Learning_System_Guides/NeoTrellis_M4_Sound_Board/code.py
59:        wave = audioio.WaveFile(f)
84:    wav = audioio.WaveFile(f)
105:            wav = audioio.WaveFile(f)
150:            wav = audioio.WaveFile(f)
```

ladyada, 2020-12-03:
@TheKitty @dherrada this is a good team up project - please start testing/adjusting some of the easier ones to get a hang of what needs to be done!

tannewt, 2020-12-07:
https://forums.adafruit.com/viewtopic.php?f=60&t=172583

dherrada, 2020-12-11:
@ladyada Sounds good. I'll try to get to it decently soon.

dherrada, 2020-12-16:
@tannewt So just to confirm, if something uses audioio.WaveFile and another audioio method like AudioOut, I should add an additional import for audiocore, and switch audioio.WaveFile to audiocore.WaveFile without changing anything else, right?

tannewt, 2020-12-16:
Correct. AudioOut is still in audioio. (RawSample is moved too I think.)

dherrada, 2020-12-17:
@TheKitty Just went through the google results for `site:learn.adafruit.com "audioio.WaveFile"` and `site:learn.adafruit.com "audioio.RawSample` and corrected the code and text in the guides to reflect WaveFile and RawSample moving.

TheKitty, 2020-12-17:
The entire Learn system is changed? Wow!

ladyada, 2020-12-17:
@TheKitty pleaes hand test a random assortment!

TheKitty, 2020-12-17:
Will do

PR #1374 (2021-01-13): Moved audioio.WaveFile -> audiocore.WaveFile
Also moved audioio.RawSample -> audiocore.RawSample

Fixes #1342
closed 2021-01-13 by dherrada