adafruit/Adafruit_Learning_System_Guides #1345: Duplicated bdf fonts in multiple packages

d-c-d, 2020-12-04:
What started as an exercise to download a few fonts for a demo, led to the discovery that there are many duplicated fonts scattered around the Adafruit_Learning_System_Guides.
  ( Note: Duplicated here means the bdf files have the same md5sum )

This could lead to maintenance issues, it also pollutes the git repository with multiple 'binary files', though I wouldn't be surprised if git was smart enough to merge the duplicates.  (That is an exercise for another day )
Git isn't smart enough to reduce disk diskspace used by all the duplicate fonts when the Adafruit_Learning_System_Guides is cloned, though the bdf fonts are only about 2.5% of the Learning Guides.

One can also see that while it appears that fonts are stored in a "fonts" subdirectory most of the time, they are sometimes stored in the main project folder.

Because of this duplication, sometimes the same font file name contains different data.

For example, when looking for Arial-Bold-12.bdf there are 9 identical copies checked into git, but one (MagTag_SpaceX/Arial-Bold-12.bdf was checked in with CRLF line terminators) while the rest were just lf line terminators.
(It is noteworthy to observe that MagTag_SpaceX/Arial-Bold-12.bdf is the **only** file with **CRLF line terminators**)

Another font, Arial-12.bdf is normally 44Kb (11 copies), but 3 copies are only 12Kb, and one is 172Kb!
( I guess it is appropriate that the scary/larger one is Halloween_Countdown_Matrix/fonts/Arial-12.bdf :-) )

One way to bring some standards to this guides might be to move the fonts into a separate repository.

This may also lead to enabling future enhancements, such as storing fonts in a binary form, instead of ascii
Since CircuitPython is still targeting microprocessors with limited space, one could foresee fonts that have been converted from the mostly HEX format into something that is half the size and faster to load.
   [(https://en.wikipedia.org/wiki/Glyph_Bitmap_Distribution_Format)]

---
```

Arial-12.bdf (41212 bytes * 11 copies) = 453332 total bytes
   CircuitPython_Pyloton/fonts
   MagTag_SpaceX
   PyBadge_Conference_Badge/fonts
   PyPortal_Calculator/fonts
   PyPortal_Electioncal_US/fonts
   PyPortal_Hackster/fonts
   PyPortal_NewNewNew/fonts
   PyPortal_OpenWeather/fonts
   PyPortal_Thingiverse/fonts
   PyPortal_Tides/fonts
   Weather_Display_Matrix/fonts

Collegiate-50.bdf (82110 bytes * 9 copies) = 738990 total bytes
   Dashblock_API/fonts
   PyPortal_Azure_Plant_Monitor/fonts
   PyPortal_Discord/fonts
   PyPortal_GithubStars/fonts
   PyPortal_LastFM/fonts
   PyPortal_LeagueLevel/fonts
   PyPortal_Reddit/fonts
   PyPortal_TwitterFollowers/fonts
   PyPortal_Youtube/fonts

Arial-Bold-12.bdf (41772 bytes * 9 copies) = 375948 total bytes
   MagTag_AdafruitQuotes
   MagTag_Cheerlights
   MagTag_CovidTracking
   MagTag_NYC_Positivity
   MagTag_Quote_Board/fonts
   MagTag_Showerthoughts/fonts
   PyPortal_HacksterStream/fonts
   PyPortal_NewNewNew/fonts
   PyPortal_UV_Index/fonts

Collegiate-24.bdf (41569 bytes * 8 copies) = 332552 total bytes
   Dashblock_API/fonts
   PyPortal_AdafruitIO/fonts
   PyPortal_LastFM/fonts
   PyPortal_LeagueLevel/fonts
   PyPortal_Reddit/fonts
   PyPortal_TwitterFollowers/fonts
   PyPortal_Youtube/fonts
   pyportal_weather_station/fonts

[... 235 lines cut ...]
Helvetica-Oblique-17.bdf (29773 bytes * 1 copies) = 29773 total bytes
   PyPortal_Email_Display

EarthHeart-26.bdf (28954 bytes * 1 copies) = 28954 total bytes
   PyPortal_Azure_Plant_Monitor/fonts

mono-bold-8.bdf (247680 bytes * 1 copies) = 247680 total bytes
   PyPortal_Tides/fonts

Roboto-Black-48.bdf (509411 bytes * 1 copies) = 509411 total bytes
   Clue_Step_Counter/fonts

Impact-24.bdf (15757 bytes * 1 copies) = 15757 total bytes
   MagTag_NextBus/fonts

mround-31.bdf (7709 bytes * 1 copies) = 7709 total bytes
   I_Voted_Pin

RobotoMono-72.bdf (170260 bytes * 1 copies) = 170260 total bytes
   PyPortal_Smart_Switch/fonts

Arial-Bold-24-Complete.bdf (745755 bytes * 1 copies) = 745755 total bytes
   PyPortal_Bitcoin/fonts

```

ladyada, 2020-12-04:
yah this is on purpose because we have auto-sub-file-folder zipping in learn guides. this keeps the assets (images, fonts, sounds) with the project. you'll also find many copies of coin.wav or whatever, we use it a lot. it is a usability decision we made - sacrifice some disk space to make projects easier to build :)

d-c-d, 2020-12-04:
Thank you @ladyada for documenting the decision, and thanks for not closing the issue just yet :-)
Often the duplication of data is a clue that something could be improved.
Agreed that there is value in keeping things together, but doesn't prevent the same font name being used in different projects, where the name is the same, but the contents are different. ( this could be a problem in the future )