adafruit/Adafruit_Learning_System_Guides #1386: Broken Links in WebUSB tutorial

ATMakersBill, 2020-12-27:
- Arduino board:  **Circuit Playground Bluefruit**

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.13

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

The WebUSB Serial & RGB demos are no longer hosted where at the URLs in the code.  
Here is the guide link: [https://learn.adafruit.com/using-webusb-with-arduino-and-tinyusb](https://learn.adafruit.com/using-webusb-with-arduino-and-tinyusb)  It was written by @TheKitty (just adding this to tag her)

The code sets the "landing page" links like this: https://adafruit.github.io/Adafruit_TinyUSB_Arduino/examples/webusb-serial Those now give 404 errors.

It looks like the demos (or very very similar demos) are hosted here at a github.io page:
[https://webusb.github.io/arduino/demos/](https://webusb.github.io/arduino/demos/)

When I tried using the code against those demos, the console demo worked and the RGB demo didn't... but I didn't spend much time on that.  I figured finding the right test page would be a good starting point.

hathach, 2020-12-27:
thank for reporting the issue, it is just fixed. Somehow the repo pages setting is changed from `master/root` to `gh-pages` (which isn't hosting the example). Please try again

ATMakersBill, 2020-12-27:
Hi Thach,
Thanks for responding... the URLs in the guide don't seem to have changed, and they are still giving 404 errors
https://adafruit.github.io/Adafruit_TinyUSB_Arduino/examples/webusb-serial
https://adafruit.github.io/Adafruit_TinyUSB_Arduino/examples/webusb-rgb

What should I have done differently?

hathach, 2020-12-28:
please add trailing `/` to the url,  github page probably change how they handle url, ` https://adafruit.github.io/Adafruit_TinyUSB_Arduino/examples/webusb-serial/` , or better have explicitly add index.html to the url as well. I have updated the example with index.html #70 and will update the learn guide as well.

ATMakersBill, 2021-01-01:
Thanks Thach

It is working well now :-)

ladyada, 2021-01-01:
@ATMakersBill if an issue is resolved, please close it so we can keep the repo tidy thx :)

closed 2021-01-02 by ATMakersBill