adafruit/Adafruit_Learning_System_Guides #1419: IRremote breaking change for "supernova_poi" sketch

ssk8, 2021-01-17:
- Arduino board: QT PY
- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.13
- Adafruit_Learning_System_Guides/Kinetic_POV/supernova_poi/ 

supernova_poi sketch works with IRremote library 2.8.0 but remote does not work with version 2.8.1 (latest as of writing)
supernova_poi.ino it still compiles and uploads but the IR-remote does not function. Downgrading IRremote library to 2.8.0 restores function

ladyada, 2021-01-17:
i think we'll just have to document that 2.8.0 is required

@PaintYourDragon plz do :)

PaintYourDragon, 2021-01-18:
That’s really baffling. Looking through their changes, 2.8.0 to 2.8.1 only seems to affect Sony remote codes, while I think the Adafruit remote uses NEC codes. But maybe I’m overlooking something. Might be worth opening an issue in the IRremote repo.

Also, had no idea the project worked on QT Py, so that’s a welcome data point, thanks! (Looks like commenting out the two AVR #includes is necessary, does that mirror your experience?)

PaintYourDragon, 2021-01-18:
Any other changes you had to make to the wiring or the code would be welcome…I’ll mention these in the guide. Thank you!

ssk8, 2021-01-19:
Yes, I had to removed the two avr import lines and switched the RECV_PIN to 3 and that worked. It looks like there is some battery test functionality which I don't think can work with the QT Py but I really didn't look into it