adafruit/Adafruit_Learning_System_Guides #1446: Circuit Playground Express TV Zapper issue with array.array

justinmbrock, 2021-02-07:
Board: Circuit Playground Express
CircuitPython Version: 6.1.0
CP Libraries used: adafruit-circuitpython-bundle-6.x-mpy-20210206.zip
Guide link: https://learn.adafruit.com/circuitpython-tv-zapper-with-circuit-playground-express/overview

This guide was written prior to CircuitPython version 4.0.1, which introduced breaking changes to, among other things, array.array, which now checks for values too large to fit (consistent with CPython). Somehow this guide worked previously, even with the array being fed values that were too large. I don't know how to fix it, and thought someone here might know how to correct the issue.
REPL output:
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
{'index': [0, 1, 1, 1, 1, 2, 2, 1, 2, 1, 2, 1, 2, 2, 2, 2, 2, 1, 1, 2, 1, 2, 1, 2, 1, 1], 'freq': 56697, 'delay': 0.21, 'repeat': 2, 'repeat_delay': 0.008, 'table': [[3968, 3993], [493, 2003], [493, 1009]]}
{'index': [0, 1, 0, 0, 0, 0, 0, 0, 2, 0, 1, 0], 'freq': 36266, 'delay': 0.29, 'repeat': 2, 'repeat_delay': 0.09, 'table': [[882, 906], [1764, 910], [882, 1790]]}
{'index': [0, 1, 2, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 2, 1, 1, 1, 1, 1, 1, 1, 1, 1, 2, 1, 1, 1, 1, 1, 1, 1, 1, 2, 1, 2, 2, 2, 2, 1, 1, 2, 1, 2, 2, 2, 2, 1, 2, 1], 'freq': 38707, 'delay': 0.27, 'repeat': 2, 'repeat_delay': 0.064, 'table': [[3332, 1665], [413, 431], [413, 1216]]}
{'index': [0, 1, 2, 1, 2, 1, 2, 2, 1, 2, 2, 2, 2], 'freq': 38706, 'delay': 0.25, 'repeat': 2, 'repeat_delay': 0.042, 'table': [[2376, 603], [1162, 625], [594, 603]]}
{'index': [0, 1, 2, 0, 1, 0, 1, 2, 0, 1, 2, 0, 1, 2, 2, 0, 0], 'freq': 39466, 'delay': 0.33, 'repeat': 2, 'repeat_delay': 0.125, 'table': [[506, 497], [506, 4074], [506, 5067]]}
{'index': [0, 1, 1, 1, 2, 2, 2, 2, 2, 1, 1, 1, 2, 2, 2, 2, 2, 2, 1, 2, 2, 2, 2, 2, 2, 1, 2, 1, 1, 1, 1, 1, 1, 1], 'freq': 38338, 'delay': 0.25, 'repeat': 2, 'repeat_delay': 0.046, 'table': [[4460, 4500], [573, 1680], [573, 567]]}
{'index': [0, 1, 1, 1, 1, 1, 0, 0, 1, 0, 1, 1, 1, 0, 1, 2, 0, 1, 1, 1, 1, 0, 1, 1, 0, 1, 0, 0, 0, 1, 0, 0], 'freq': 38706, 'delay': 0.24, 'repeat': 2, 'repeat_delay': 0.04, 'table': [[310, 1694], [310, 700], [310, 39767]]}
{'delay': 0.23, 'freq': 38705, 'index': [0, 1, 1, 2, 2, 2, 2, 2, 2, 1, 1, 1, 2, 1, 2, 2, 2, 3, 1, 1, 2, 2, 2, 2, 2, 2, 1, 1, 1, 2, 1, 2, 2, 2, 3, 1, 1, 2, 2, 2, 2, 2, 2, 1, 1, 1, 2, 1, 2, 2, 2, 1], 'table': [[8370, 4175], [516, 1576], [516, 553], [516, 22213]]}
{'delay': 0.3, 'freq': 38339, 'index': [0, 1, 1, 2, 1, 1, 1, 1, 1, 2, 2, 1, 2, 2, 2, 2, 2, 1, 1, 1, 2, 1, 1, 1, 1, 2, 2, 2, 1, 2, 2, 2, 2, 3, 4, 5, 4, 1], 'table': [[8946, 4488], [547, 590], [547, 1709], [547, 39729], [8946, 2253], [547, 95603]]}
Traceback (most recent call last):
  File "code.py", line 67, in <module>
OverflowError: value must fit in 2 byte(s)

Press any key to enter the REPL. Use CTRL-D to reload.```

ladyada, 2021-02-07:
wanna try fixing? you'd have to set the values to maximum 65535!

justinmbrock, 2021-02-08:
Easy enough to change. I also spotted a module that changed since it was written and updated the code for that as well.

PR #1447 (2021-02-08): Updates to the Circuit Playground Express TV Zapper
Some updates to the Circuit Playground Express TV Zapper code to bring it in line with CircuitPython 6.1.0. The changes to code.txt closes #1446 .

Changes:
codes.txt - limited the table values to 65535 so as to not overflow the 2 byte (half word) limit for pulseio.PulseOut
cpx_main.py - added ```import pwmio```, the module was originally part of pulseio and is now separate
gemma_main.py - added ```import pwmio```... same as above

I tested the changes on a Circuit Playground Express and successfully turned off an Insignia TV.
closed 2021-02-08 by ladyada