adafruit/Adafruit_Learning_System_Guides #1450: Maybe a typo

sutha064, 2021-02-13:
- Arduino board:  Arduino UNO

- Arduino IDE version 1.8.13

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):
https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/master/GP2Y1014AU0F_demos/GP2Y1014AU/GP2Y1014AU.ino

      void loop() {
          float output_voltage, dust_density;
          
          digitalWrite(ledPin, LOW); // power on the LED
          delayMicroseconds(280); // Wait 0.28ms according to DS
          // take analog reading
          output_voltage = analogRead(dustPin); 
          delay(1);
          
          digitalWrite(ledPin, HIGH); // turn the LED off

Here 'ledPin' is set to LOW while it is commented to power it on and vice versa. Copied from official repo, check it on link above the code.

ladyada, 2021-02-13:
please explain specifically why this example is wrong, with hardware testing and analysis. are you just guessing?

closed 2021-02-13 by ladyada

sutha064, 2021-02-14:
Hello, correct me if I am wrong, dust sensor has mainly two components, an LED and a sensor. The LED will light up and then an analog reading is taken. Then the LED is turned off. Your comments on the side of the code also suggest the same.
    digitalWrite(ledPin, LOW); // power on the LED
But isn't the code doing exactly the opposite?

ladyada, 2021-02-14:
pretty sure that's wrong / guessing - feel free to find real evidence that we're mistaken! :)

sutha064, 2021-02-14:
what? why so rude? do you think I am trying to prove you are wrong and I am right? I am just confused and can you rather explain how the code functions to the beginner who is trying to learn do I have to find it myself? How can a senior electrical engineer respond like this? I mean I am just learning and everyone makes mistakes. Do correct my understanding kindly!