adafruit/Adafruit_Learning_System_Guides #1523: Feed key must contain lower case English letters, numbers, and dash

jeremykeen, 2021-04-24:
Error when running [scoreboard code](https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/master/Matrix_Portal_Scoreboard/matrix_portal_scoreboard.py) on Matrix Portal due to adafruit IO feed name validation.  The error is "Feed key must contain lower case English letters, numbers, and dash." The code uses `scores-group.red-team-score-feed` for the feed name.  The [validation code](https://github.com/adafruit/Adafruit_CircuitPython_AdafruitIO/blob/master/adafruit_io/adafruit_io.py) line 37 throws the error due to the `.`

Tested moving the feeds to the default group and removing the `scores-group.` This resolves the issue.

brentru, 2021-05-03:
This issue is addressed in https://github.com/adafruit/Adafruit_CircuitPython_AdafruitIO/pull/73

brentru, 2021-05-05:
@jeremykeen  We've patched this issue and released a new version of the Adafruit IO CircuitPython library, this issue should no longer persist.

Please delete the `adafruit-io` library from your CIRCUITPY drive and replace it with the latest version of the library: https://github.com/adafruit/Adafruit_CircuitPython_AdafruitIO/releases/tag/5.2.5