adafruit/Adafruit_Learning_System_Guides #1530: Implement auto-screenshotter!

ladyada, 2021-05-01:
from @FoamyGuy:

> The current version of the scripts are located here: https://github.com/FoamyGuy/CircuitPython_Library_Screenshot_Maker/tree/main/python_generator

@jepler you were taking a crack at this? i'd like the png/image to be commited into the project's folder

kevinjwalters, 2021-05-04:
Is this to produce the listing of library files for one or more programs in a Guide?

I asked about this back in November but assumed JavaScript (fed with generated meta-data about the program) would be the more flexible solution.

> [21:39] kjw: kattni For the images of file system trees for libraries in the guides, has there been any discussion on generating those with some javascript magic? That would make it easier and quicker to update them - it could generate both Mac and Windows look

Was (direct) image generation chosen for compatibility with producing pdfs?

FoamyGuy, 2021-05-04:
@kevinjwalters I'm glad you mentioned something! I could have sworn I saw the idea go by in the discord but couldn't remember when or who said it.

Your comment floated around in my head for a while, and I did create an HTML / JS version first which was a bit more manual. The required libraries were entered by the user, and it only rendered on the screen, leaving the user to screenshot it however they wanted. I may have misinterpreted your original idea though but this is essentially what came to mind when you originally proposed it on discord.

We switched it up to Python / PIL  directly generating the images in order to make it more automated (the new version runs on all the Learn Guide Projects at once) and to make it easier to run from actions / CI. It may be possible to execute JS in that context (I'm not certain) but I have much more experience with Python so that seemed like an easier route.