adafruit/Adafruit_Learning_System_Guides #1602: Hallowing Jump Scare Trap needs updated

FoamyGuy, 2021-07-10:
In the adafruit Jump Scare Trap it tries to make use of `displayio.Sprite` but this class does not exist in modern versions of circuitpython / displayio.

https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/526c7ed79795ea1b2d4b033004f79892a084c8ad/HalloWing_Jump_Scare_Trap/hallowing_jump_scare_trap.py#L99

I think it needs to be modified to use TileGrid instead of Sprite. 

I'll look into making these changes, wanted to get an issue created so I don't forget about it.

FoamyGuy, 2021-07-10:
HalloWing_Tour_Guide also needs this same update

https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/526c7ed79795ea1b2d4b033004f79892a084c8ad/HalloWing_Tour_Guide/code.py#L69

perhaps more as well. I'll keep adding comments here as I find them and then come back and do a PR to fix all at once.

lesamouraipourpre, 2021-08-06:
This should probably be closed. It has been incorporated into the tracking issue #1747

closed 2021-08-06 by FoamyGuy