adafruit/Adafruit_Learning_System_Guides #1603: Remove max_size from CircuitPython examples that use Group

lesamouraipourpre, 2021-07-10:
The `max_size` parameter on `displayio.Group` is currently ignored and is (hopefully) going to be removed before 7.0.0 final.
The libraries are having this parameter removed from them - https://github.com/adafruit/circuitpython/issues/4959

This issue is intended as a tracking issue as the Learn Guide code is worked through
`git grep "Group.*max_size"` gives
```
AHT20_OLED/code.py:splash = displayio.Group(max_size=8)
Baudot_TTY/baudot_tty_GUI.py:screen = displayio.Group(max_size=5)
Baudot_TTY/baudot_tty_GUI.py:message_group = displayio.Group(max_size=5, scale=1)
Buckaroo_Plant_Care_Bot/buckaroo_plant_care_bot.py:clue_display = displayio.Group(max_size=4)
Buckaroo_Plant_Care_Bot/buckaroo_plant_care_bot.py:text_group = displayio.Group(max_size=3, scale=3)
CLUE_Altimeter/clue_altimeter.py:splash = displayio.Group(max_size=4)
CLUE_Altimeter/clue_altimeter.py:alti_readouts = displayio.Group(max_size=2, scale=6)
CLUE_BBQ/clue_bbq.py:homescreen_screen = displayio.Group(max_size=3)
CLUE_BBQ/clue_bbq.py:temperatures_screen = displayio.Group(max_size=2)
CLUE_BBQ/clue_bbq.py:text_group = displayio.Group(max_size=8, scale=1)
CLUE_Egg_Drop/clue_egg_drop.py:splash = displayio.Group(max_size=3)
CLUE_Egg_Drop/clue_egg_drop.py:text_group = displayio.Group(max_size=1, scale=2, x=10, y=220)
CLUE_Hand_Wash_Timer/clue_hand_wash_timer.py:clue_display = displayio.Group(max_size=4)
CLUE_Hand_Wash_Timer/clue_hand_wash_timer.py:text_group = displayio.Group(max_size=5, scale=1)
CLUE_I_Ching/clue_iching.py:hexagram = displayio.Group(max_size=1, x=60, y=15, scale=10)
CLUE_Light_Painter/code.py:        group = displayio.Group(max_size=7)
CLUE_Metal_Detector/clue-metal-detector.py:screen_group = Group(max_size=4 + 4 + 1)
CLUE_Metronome/clue_metronome.py:screen = displayio.Group(max_size=11)
CLUE_Milk_Checker/code.py:splash = displayio.Group(max_size=5)
CLUE_Rock_Paper_Scissors/advanced/rps_display.py:    disp.show(Group(max_size=1))
CLUE_Rock_Paper_Scissors/advanced/rps_display.py:        round_choice_group = Group(max_size=3)
CLUE_Rock_Paper_Scissors/advanced/rps_display.py:        s_group = Group(scale=3, max_size=1)
CLUE_Rock_Paper_Scissors/advanced/rps_display.py:            intro_group = Group(max_size=7)
CLUE_Rock_Paper_Scissors/advanced/rps_display.py:                s_group = Group(scale=3, max_size=1)
CLUE_Rock_Paper_Scissors/advanced/rps_display.py:        playerlist_group = Group(max_size=self.max_players * 2)
CLUE_Rock_Paper_Scissors/advanced/rps_display.py:        gs_group = Group(max_size=4)
CLUE_Rock_Paper_Scissors/advanced/rps_display.py:        scores_group = Group(max_size=len(pla))
CLUE_Rock_Paper_Scissors/advanced/rps_display.py:            error_group = Group(max_size=error_tot + 1)
CLUE_Rock_Paper_Scissors/advanced/rps_display.py:            pvp_group = Group(max_size=3)
CLUE_Rock_Paper_Scissors/advanced/rps_display.py:                s_group = Group(scale=2, max_size=2)  # Audio is choppy at scale=3
CLUE_Rock_Paper_Scissors/simple/clue-simple-rpsgame.py:screen_group = Group(max_size=len(choices) * 2 + 1 + 1)
CLUE_Sensor_Plotter/plotter.py:        g_plot_grid = displayio.Group(max_size=3)
CLUE_Sensor_Plotter/plotter.py:        g_background = displayio.Group(max_size=3+len(plot_y_labels))
CLUE_Sensor_Plotter/plotter.py:        main_group = displayio.Group(max_size=3)
CPB_AMS_Gizmo_BLE/cpb_ams_gizmo_ble.py:group = displayio.Group(max_size=20)
CPB_ANCS/code.py:group = displayio.Group(max_size=3)
CircuitPython_Display_Text/background_color_example.py:main_group = displayio.Group(max_size=10)
CircuitPython_Display_Text/background_tight_example.py:main_group = displayio.Group(max_size=10)
CircuitPython_Display_Text/base_alignment_test.py:main_group = displayio.Group(max_size=10)
CircuitPython_Display_Text/color_example.py:main_group = displayio.Group(max_size=10)
CircuitPython_Display_Text/colormask_example.py:main_group = displayio.Group(max_size=10)
CircuitPython_Display_Text/display_text_anchored_position.py:main_group = displayio.Group(max_size=10)
CircuitPython_Display_Text/display_text_anchored_position.py:text_group = displayio.Group(max_size=9)
CircuitPython_Display_Text/font_example.py:main_group = displayio.Group(max_size=10)
CircuitPython_Display_Text/line_spacing_example.py:main_group = displayio.Group(max_size=10)
CircuitPython_Display_Text/padding_example.py:main_group = displayio.Group(max_size=10)
CircuitPython_Display_Text/scale_example.py:main_group = displayio.Group(max_size=10)
CircuitPython_Flying_Toasters/code.py:    group = displayio.Group(max_size=10)
CircuitPython_JEplayer_mp3/code.py:    board.DISPLAY.show(displayio.Group(max_size=1))
CircuitPython_JEplayer_mp3/code.py:        self.group = displayio.Group(max_size=4)
CircuitPython_JEplayer_mp3/code.py:    scene = displayio.Group(max_size=len(labels) + 1)
CircuitPython_JEplayer_mp3/icons.py:        self.group = displayio.Group(max_size=n)
CircuitPython_PyPaint/code.py:        self._cursor_grp = displayio.Group(max_size=1)
CircuitPython_PyPaint/code.py:        self._splash = displayio.Group(max_size=5)
CircuitPython_Pyloton/pyloton.py:        self.splash = displayio.Group(max_size=25)
CircuitPython_RGBMatrix/fruit.py:g = displayio.Group(max_size=3)
CircuitPython_RGBMatrix/life.py:g1 = displayio.Group(max_size=3, scale=SCALE)
CircuitPython_RGBMatrix/life.py:g2 = displayio.Group(max_size=3, scale=SCALE)
CircuitPython_RGBMatrix/scroller.py:g = displayio.Group(max_size=2)
CircuitPython_RGBMatrix/scroller.py:l1 = displayio.Group(max_size=linelen)
CircuitPython_RGBMatrix/scroller.py:l2 = displayio.Group(max_size=linelen)
CircuitPython_RPN_Calculator/code.py:        g = displayio.Group(max_size=7)
CircuitPython_Scrolling_Clouds/code.py:    group = displayio.Group(max_size=10)
CircuitPython_SharpDisplay_Displayio/code.py:blm_group = displayio.Group(max_size=10)
[... 104 lines cut ...]
RGB_Matrix_Slot_Machine/code.py:g = displayio.Group(max_size=3)
Say_Their_Names_LEDMatrix/code.py:g = displayio.Group(max_size=10)
Smart_Alarm_Clock/code.py:splash = displayio.Group(max_size=13)
TFT_Gizmo_Snowglobe/snowglobe_fancy.py:flakes = displayio.Group(max_size=NUM_FLAKES)
TFT_Gizmo_Snowglobe/snowglobe_simple.py:flakes = displayio.Group(max_size=NUM_FLAKES)
Tilemap_Game_With_CircuitPython/code.py:group = displayio.Group(max_size=33)
Tilemap_Game_With_CircuitPython/code.py:    splash = displayio.Group(max_size=4)
Tilemap_Game_With_CircuitPython/code.py:    text_group = displayio.Group(max_size=8, scale=1, x=14, y=18)
Vertical_Garden_Barometer/code.py:clue_display = displayio.Group(max_size=4)
Vertical_Garden_Barometer/code.py:text_group = displayio.Group(max_size=5, scale=1)
Voice2Json_Edge_Detection/demo.py:splash = displayio.Group(max_size=10)
Weather_Display_Matrix/openweather_graphics.py:        splash = displayio.Group(max_size=1)
Weather_Display_Matrix/openweather_graphics.py:        self.root_group = displayio.Group(max_size=15)
Weather_Display_Matrix/openweather_graphics.py:        self._icon_group = displayio.Group(max_size=1)
Weather_Display_Matrix/openweather_graphics.py:        self._text_group = displayio.Group(max_size=5)
Weather_Display_Matrix/openweather_graphics.py:        self._scrolling_group = displayio.Group(max_size=1)
azure_gfx_helper.py:        root_group = displayio.Group(max_size=20)
azure_gfx_helper.py:        self._icon_group = displayio.Group(max_size=1)
azure_gfx_helper.py:        self._text_group = displayio.Group(max_size=6)
pyportal_pet_planter/code.py:splash = displayio.Group(max_size=15)
pyportal_weather_station/weatherstation_helper.py:        root_group = displayio.Group(max_size=20)
pyportal_weather_station/weatherstation_helper.py:        self._icon_group = displayio.Group(max_size=1)
pyportal_weather_station/weatherstation_helper.py:        self._text_group = displayio.Group(max_size=8)
```
If the associated Learn Guide text needs updated I will make a comment on any created PRs. Is there a better way to do this @kattni or @makermelissa ?

kattni, 2021-07-12:
@lesamouraipourpre Unfortunately, this is pretty much the only way to do it. Thank you for making this list! This update needs to be made, and updating the code is a great first step. Not all guides will have associated walk-throughs or text requiring changes, but guarantee that some of them will.

Perhaps @FoamyGuy is up for assisting with this on the Learn guide side of things?

lesamouraipourpre, 2021-07-12:
Not a problem, I will try to start working through these. A potential issue noticed on @FoamyGuy 's stream on Saturday was that some Learn Guides were written for CircuitPython 3 (and possibly earlier) and may need other changes.

As I don't have the hardware for most of the guides, I will focus on the `max_size` change and try and report any other issues.

FoamyGuy, 2021-08-01:
I am going to start working on this list from the bottom moving upwards to avoid accidentally duplicating any of @lesamouraipourpre 's work

lesamouraipourpre, 2021-08-02:
Sounds Good. I'll continue downwards.

lesamouraipourpre, 2021-08-03:
@FoamyGuy I've modified the `grep` search to look for `max_glyphs` as well - it's added a few more to the list.

lesamouraipourpre, 2021-08-12:
All current Learn guides have had `max_size` and `max_glyphs` usage removed.

closed 2021-08-12 by lesamouraipourpre

ladyada, 2021-08-12:
wooohoo! thank you :)