adafruit/Adafruit_Learning_System_Guides #1847: Copyright and License notices on MACROPAD example?

rpavlik, 2021-09-28:
The MacroPad example is great to customize and share, however, I hesitate to do so without a clear copyright and license notice. I see that some other samples have SPDX tags which would be great. For my own macro configs I'm using CC0-1.0. I'd be happy with pretty much any permissive license you already use to be placed on https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/Macropad_Hotkeys/code.py and associated files

I use the https://reuse.software tool to ensure (in CI) that my projects all have complete and correct copyright/licensing statements for each file. I think maybe some of your circuitpython libraries do too? Might be harder in this repo because of the binary assets and the desire to make small bundles, but steps in that direction would be appreciated.

Thanks for your great resources (and hardware!)

TheKitty, 2021-09-28:
I agree with @rpavlik that labeling has been inconsistent and Adafruit should establish a methodology for license information.

rpavlik, 2021-09-29:
It's actually pretty good on the library repositories, I think: I'm pretty sure at least some have the reuse CI check enabled. It's just this one, which ironically is the one most likely to be modified. (Many of my Adabox project downloads do actually have spdx tags for the code too, though less often for the assets, this macro pad one is the first in a while that I remember missing that data.)

closed 2021-09-29 by TheKitty

rpavlik, 2021-09-30:
It looks like the file I linked is still missing notices?

TheKitty, 2021-09-30:
We have just started a project to go through the entire Learn repo and add
license information. It'll take a bit of time, though.

rpavlik, 2021-10-01:
Oh awesome, thanks. No worries!