adafruit/Adafruit_Learning_System_Guides #1875: Substitute Trinket M0 (3V?) for obsolete Trinket (5V)

NohWayJose, 2021-10-03:
I am following the Instructable for making pixel poi from here (https://learn.adafruit.com/genesis-poi-dotstar-led-persistence-of-vision-poi/preparing-images?view=all)

The project recommendation is to use the 5V Adafruit Trinket (https://www.adafruit.com/product/1501)  but this is deprecated
The Adafruit  recommended replacement is the Trinket M0 (https://www.adafruit.com/product/1501)

**Does anyone know if this is a compatible replacement** before I place an order for two of them?

**If it is substitutable, are there any gotchas _(hardware or software)_** I need to be aware of?

ladyada, 2021-10-03:
hiya for tech support questions, please post on discord or in the adafruit forums :) thanxs!

closed 2021-10-03 by ladyada

NohWayJose, 2021-10-03:
Sorry, will do. Do you want me to delete and recreate (move?) it or is this a request for next time?