adafruit/Adafruit_Learning_System_Guides #1939: For BLE connect and UART

LeonLyu, 2021-12-07:
Hi guys I am stuck with the part of sending information from adafruit clue to my phone by BluefruitConnect, I am trying to make a timer and the board needs to send time count down to my phone(like 30, 29, 28....,1) I am not sure what is the code of sending sequence to the UART, could anyone help me?

dglaude, 2022-01-04:
I believe you should describe this in the support forum: https://www.adafruit.com/support
Or if you could try Discord https://adafru.it/discord and find the help with channel that match your language of choice (arduino or circuitpython).

Maybe you should close this issue.

dhalbert, 2022-01-04:
Yes, please use one of those areas for support, thanks.

closed 2022-01-04 by dhalbert