adafruit/Adafruit_Learning_System_Guides #1941: Missing 2022.h file for "Shake Away 2020 with MatrixPortal By Phillip Burgess"

dglaude, 2021-12-08:
The learn guide at
https://learn.adafruit.com/matrixportal-shake-away-2020 would need an update since 2022 approach.

And we need a 2022.h with the text version of a bitmap for 2022 to be used in adafruit/Adafruit_Learning_System_Guides/tree/main/2020_shake

And a few other change to replace everywhere in the code 2021 by 2022 and 2020 by 2021. 

Thanks

ladyada, 2021-12-09:
yes! please submit a PR :) happy new year

caternuson, 2021-12-09:
Could also make code more generic at same time? Like just have a single header file with two bitmaps defined `bitmap_outgoing[] = ` and `bitmap_incoming[] = ` or similar. And then that's all that would need updating.

Could keep all the `YEAR.h` files, like `2020.h` and `2021.h` , in the repo for reference? And just copy-paste whatever years were wanted from them into the single header file.

dglaude, 2021-12-09:
The problem is that the guide does not really describe how to get from a bitmap to the table full of hex value.
Even if I could figure it out... having the original author do the update that would be best you also want to update the guide and publish new UF2 files, so there more to it than what an outsider can do.

I also added a message on the guide page to make sure.

ladyada, 2021-12-09:
@PaintYourDragon any hints on how @dglaude or others can update it?

PaintYourDragon, 2021-12-14:
Guide and code are updated. Guide includes new UF2s (2021–2022 switch), code (download ZIP) has several extra .h files through the 2025–2026 switch and is somewhat more configurable. Also added link in guide for the Python tool that does the PNG-to-dot-h conversion: https://github.com/adafruit/Adafruit_Media_Converters

closed 2021-12-14 by PaintYourDragon