adafruit/Adafruit_Learning_System_Guides #1945: Hi / Lo times are NOT being displayed expected in the "MagTag Tides Viewer" tutorial

wavesailor, 2021-12-10:
I found a bug in the following code: https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/MagTag_Tides/code.py

Currently the Hi / Lo times are NOT being displayed expected - They are just missing. Everything else works.

I'm using the latest library and Circuit Python for MagTag
```
adafruit-circuitpython-adafruit_magtag_2.9_grayscale-en_US-7.0.0.uf2
adafruit-circuitpython-bundle-7.x-mpy-20211210.zip
```

wavesailor, 2021-12-10:
**I found the solution.**

this line of code `        hilo_times[i].hidden = False` needs to be added to the end of this function `def show_hilo():`

Then it works as it should.

ladyada, 2021-12-10:
great! wanna do a PR? :)

FoamyGuy, 2021-12-10:
I spoke with this user on Discord as well. I can PR the fix for this and one other minor issue they reported this afternoon.

caternuson, 2021-12-10:
Weird. Wonder how this ever worked? Maybe somehow got away with it in 6.x?

@wavesailor Thanks for finding this.

@FoamyGuy Thanks for PR. I'd suggest adding the visible setting here along with the icon:
https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/0db1ba2c12a843a797793b09b731a7e726cb8c52/MagTag_Tides/code.py#L232-L233

PR #1946 (2021-12-11): Magtag project fixes
Resolves: #1945 

@caternuson These are the fixes that we discussed on Discord.

Tide texts were not getting `hidden = False` previously. I think one or more of the libraries got updated at some point and it made this behavior different than it was originally. I ran the Tides project a few months back and the text was visible at that time.

This PR also scoots the bottom line of text a little bit higher in the sports schedule project. Previously the bottom 1 or 2 pixels was partially off the screen.

Also ran black formatting which added a few spaces that were otherwise un [... cut]
closed 2021-12-11 by FoamyGuy