adafruit/Adafruit_Learning_System_Guides #1975: /CircuitPython_Essentials/Pin_Map_Script/code.py error on RPI Zero W

nightmechanic, 2022-01-12:
- board.board_id:  RASPBERRY_PI_ZERO_W

Adafruit-Blinka==6.18.0
adafruit-circuitpython-busdevice==5.1.1
adafruit-circuitpython-max31855==3.2.10
Adafruit-PlatformDetect==3.19.2
Adafruit-PureIO==1.1.9
adafruit-python-shell==1.3.3

- I get the following error when trying to run the pin_map_script code.py:

`File "/home/pi/Pin_Map_orig.py", line 6, in <module>
    for pin in dir(microcontroller.pin):
AttributeError: module 'microcontroller' has no attribute 'pin'`

I got the script to work by dropping the '.pin' from microcontroller:

`
"""CircuitPython Essentials Pin Map Script"""
import microcontroller
import board

board_pins = []
for pin in dir(microcontroller):
    if isinstance(getattr(microcontroller, pin), microcontroller.Pin):
        pins = []
        for alias in dir(board):
            if getattr(board, alias) is getattr(microcontroller, pin):
                pins.append("board.{}".format(alias))
        if len(pins) > 0:
            board_pins.append(" ".join(pins))
for pins in sorted(board_pins):
    print(pins)
`
What am I missing?

TheKitty, 2022-01-12:
@tannewt is this a syntax error or possible broadcom def issue?

closed 2022-03-04 by kattni

makermelissa, 2022-03-04:
This script was not originally designed to run on Blinka. However, as of Blinka 7.1.0, it now works without modification.