adafruit/Adafruit_Learning_System_Guides #1987: 3 issues with the HalloWing GPS Tour Guide project

jeremyharper, 2022-01-22:
I've been trying to get [this project](https://learn.adafruit.com/gps-tour-guide) working and I ran into 3 bugs with the current version of the sample code. Apologies for not just putting in a pull request but I am super rusty at using git:

https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/HalloWing_Tour_Guide/code.py

1. On line 24, the timeout needs to be between 1 and 100 seconds (looks like this is sending it in ms, should change timeout to 3 instead of 3000)
2. Lines 43-45, 81, and 84 throw a TFT_Backlight in use error. I had luck replacing it with board.DISPLAY.brightness = 0 and board.DISPLAY.brightness = 1 as appropriate
3. I kept getting memory errors whenever it would attempt to play a wav file. Removing the debouncer library (reverting to the old code from November 12, 2018 for the switch instead) made it play fine.

I can't imagine that the debouncer library is THAT heavyweight of a library but it seems to be just enough to run the HalloWing out of memory.

ladyada, 2022-01-26:
@dastels take a look?

dastels, 2022-01-26:
1. You're right.  I'd make the default 3.0 just to be clearer that it's a float.
2. I'm not sure why it would already be in use. Unless there's been changes in displayio that grabs it. Seems unlikely since it's board specific and not display specific.
3. What version of CircuitPython are you using?  It simply may not fit on the M0 with more recent versions of CP and the libraries. It looks like it was originally written for CP3 back when I wrote it. (late 2018/early 2019).

dastels, 2022-01-26:
I checked busio and when this project was written the timeout was milliseconds as an int.  I suspect that's the whole reason for the problems: itr was written for a previous version of CP and a lot has changed since then.

ladyada, 2022-01-26:
@dastels ok wanna try it and see? if it doesnt fit it should be adaptable to an m4?

jeremyharper, 2022-01-26:
I think you're right--I was using the latest version, CircuitPython 7.11.

That's probably the cause of issue 2 as well. It looks like according to [this](https://github.com/adafruit/circuitpython/issues/1660#issuecomment-473793461) that the backlight control was moved into a different object at some point

dastels, 2022-01-26:
I'm having trouble finding the correct bundle. Only the latest 3.x version is linked. The appropriate CP doesn't seem to include displayio.

ladyada, 2022-01-26:
right you need to move it to CP 7, we dont generate MPYs for earlier versions anymore