adafruit/Adafruit_Learning_System_Guides #2115: Getting a request.py error in the code for the "MagTag Lists From Google Spreadsheets" tutorial

wavesailor, 2022-03-21:
I'm getting an error running the following code: https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/MagTag_Google_Sheets/weekly_planner/code.py
```
Connecting to AP MyWiFI
Updating time
Getting time for timezone America/New_York
Updating tasks
Retrieving data...Traceback (most recent call last):
  File "code.py", line 101, in <module>
  File "adafruit_portalbase/network.py", line 478, in fetch
  File "adafruit_requests.py", line 769, in get
  File "adafruit_requests.py", line 758, in request
  File "adafruit_requests.py", line 702, in request
  File "adafruit_requests.py", line 378, in close
ValueError: invalid syntax for integer with base 16

Code done running.
```

I use `wget` on my Pi  to retrieve my URL just fine, but it does not work on the MagTag

I'm using the latest libraries and Circuit Python for MagTag
```
adafruit-circuitpython-adafruit_magtag_2.9_grayscale-en_US-7.2.3.uf2
adafruit-circuitpython-bundle-7.x-mpy-20220321.zip
```

anecdata, 2022-03-21:
Google may have changed up the URLs and something may be getting lost in a redirect. Can you test with curl, something like:
```py
curl -iL -A "Adafruit CircuitPython" --http1.1 "https://docs.google.com/spreadsheets/d/e/XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/pub?output=tsv"
```
and see what the sequence of URLs is.

anecdata, 2022-03-21:
Looks to me like Requests will do redirects _within the same host_:
https://github.com/adafruit/Adafruit_CircuitPython_Requests/blob/270565665ada26fe8d7a99a3cb5941b452444471/adafruit_requests.py#L737 but Google does the `307` to a new subdomain & domain.