adafruit/Adafruit_Learning_System_Guides #2162: REPL spits out errors in the macropad-hotkeys project

imavroukakis, 2022-05-16:
- Arduino board:  Adafruit MacroPad RP2040 
- Arduino IDE version:  N/A using Circuit Python
- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):

Using CircuitPython 7.2.5, I am playing around with https://learn.adafruit.com/macropad-hotkeys/project-code . I have noticed that the console spits out a lot of errors when it loads the macros but everything seems to work fine

```
ERROR in ._mac-safari.py
Traceback (most recent call last):
  File "code.py", line 101, in <module>
ImportError: no module named '/macros/'
ERROR in ._media.py
Traceback (most recent call last):
  File "code.py", line 101, in <module>
ImportError: no module named '/macros/'
ERROR in ._numpad.py
Traceback (most recent call last):
  File "code.py", line 101, in <module>
ImportError: no module named '/macros/'
```

ladyada, 2022-05-16:
you probably didnt Download Bundle to get all the files?
[image]
[image]

imavroukakis, 2022-05-16:
Sure did!
[image]

ladyada, 2022-05-16:
._ files are macos 'extras' made by your OS, you could delete them using command line - they're confusing the script but like you said work fine

imavroukakis, 2022-05-16:
ah thanks, I thought I squashed those when I followed the setup instructions! Apologies for that.

ladyada, 2022-05-16:
all good! :) if ya want you could submit a PR for the script to ignore files that start with `._` !

closed 2022-05-16 by imavroukakis