adafruit/Adafruit_Learning_System_Guides #2186: ArduinoISP.ino does not properly handle word-based address for EEPROM read/write

x0urc3, 2022-06-14:
**Steps to reproduce**
Reading and writing EEPROM using Adafruit ArduinoISP.ino will cause Bad EEPROM data. The official Arduino [ArduinoISP.ino](https://github.com/arduino/arduino-examples/tree/main/examples/11.ArduinoISP/ArduinoISP) does have similar issue. See avrdudes/avrdude/issues/990
 
**Possible solution**
The read and write EEPROM function does not handle word-based address. The high address byte is 0x00 in both function. 
https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/0090edd636158f91e5356d7806f3f21cea0c9cde/ArduinoISP/ArduinoISP/ArduinoISP.ino#L550-L557
https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/0090edd636158f91e5356d7806f3f21cea0c9cde/ArduinoISP/ArduinoISP/ArduinoISP.ino#L487-L496

TheKitty, 2022-06-14:
Please submit a pull request if at all possible.

mcuee, 2022-06-14:
FYI, Arduino's example has the fix back in Dec 2011.
https://github.com/arduino/arduino-examples/commit/ea98fc8f98713fe8b896e2bd813a1fa409419deb