adafruit/Adafruit_Learning_System_Guides #2223: File has misleading name

mew-cx, 2022-07-17:
Regarding the file [CircuitPython_Essentials/CircuitPython_Logger_Boot/code.py](https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/CircuitPython_Essentials/CircuitPython_Logger_Boot/code.py)...
It really seems the file should be named `boot.py` in the repo, rather than `code.py`.

Its companion file [CircuitPython_Essentials/CircuitPython_Logger/code.py](https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/CircuitPython_Essentials/CircuitPython_Logger/code.py) is also (but appropriately) named `code.py`.

Admittedly, the directions at [CircuitPython Storage](https://learn.adafruit.com/circuitpython-essentials/circuitpython-storage) do say to save the file on the device as `boot.py`, but it just seems unnecessarily confusing not to have the file properly named (especially when the name of the file is critical for the example to work!)

I'm glad to provide a PR for this simple single-file renaming.

kattni, 2022-07-25:
@mew-cx Thanks so much for reporting this. A pull request would be great! Please let me know if you run into any issues, or need assistance. When you submit the PR, please link it here and tag me. Thank you!

mew-cx, 2022-07-29:
@kattni The renaming PR is #2229 

There are (at least) two places in the documentation that refer to the file.  They probably need some editing, since boot.py no longer needs to be renamed:
- https://learn.adafruit.com/circuitpython-essentials/circuitpython-storage
- https://learn.adafruit.com/adafruit-gemma-m0/circuitpython-storage-2
I could edit the docs, but I don't know where the live in the repo.  If you'd like me to edit, please point them out.
HTH
-- mew

dglaude, 2022-07-29:
To the best of my knowledge...
The learn guide text/image are unfortunately not on GitHub but on a separate system.
It is available to Adafruit worker and external contributor.
But don't worry Kattni is an expert and produce/update many learn guides.
If one day you want to create a learn guide for one of your project it is totally possible and you get introduction/help on the system, there is even a dedicated channel on discord to discuss with other writer/translator of guide.

PR #2229 (2022-08-08): Rename file to its intended name of "boot.py"
Rename file in the repo to `boot.py`, instead of requiring it to be renamed manually.
Fixes #2223
closed 2022-08-08 by kattni