adafruit/Adafruit_Learning_System_Guides #2266: MagTag weather: Open weather API limitations to new accounts

Neradoc, 2022-09-03:
Hi,
In the MagTag open weather project, it seems that the onecall APIs (get_forecast) are not accessible to a new free account anymore and require to subscribe to the onecall service, which is "free" but requires registering a card to pay-per-call above the allotted free ones. The default free accounts only include the current weather and 3-hours forecast, not the daily. <https://openweathermap.org/price#weather>

Using a new account, after making sure the API key is active, as it works with the calls to the free Geocoding API, I get the following response from the site:
```py
{'message': 'Invalid API key. Please see http://openweathermap.org/faq#error401 for more info.', 'cod': 401}
```

Which is described as what the API returns when not authorized to access in the mentioned link (other causes being ruled out).
> You are using a Free subscription and try requesting data available in other subscriptions . For example, 16 days/daily forecast API, any historical weather data, Weather maps 2.0, etc). Please, check your subscription in your personal account.

This subsequently leads to:
```py
Traceback (most recent call last):
  File "code.py", line 280, in <module>
  File "code.py", line 92, in get_forecast
KeyError: daily
```

When I use the account I created way back when the guide was new it allows me to access the onecall API and the code runs as expected. I think the guide has to be revised to mention that harsh limitation, or another API has to be found.

Note that the free tier has "3-hour" forecast over 5 days (`https://api.openweathermap.org/data/2.5/forecast?&lat=&lon=`) so we could loop through that to find the min-max temperatures over a day for example, but the rest would be more complicated.

tlyu, 2022-09-03:
> This subsequently leads to:
> 
> ```python
> Traceback (most recent call last):
>   File "code.py", line 280, in <module>
>   File "code.py", line 92, in get_forecast
> KeyError: daily
> ```

I think the code should also check the HTTP status code, and give a more useful error here, instead of trying to look up JSON keys as if there were always a successful response.

caternuson, 2022-09-06:
Was the 2.5 API being used in all cases? I ran into something similar here:
https://forums.adafruit.com/memberlist.php?mode=viewprofile&u=345839

The key I created way back when writing the guide still works with 2.5, but not 3.0. As long as the 2.5 API is used, the guide still works as shown (with that key at least).

I did create a new key and verified that it still did not work with the 3.0 API. The 3.0 API is just not as free anymore. So it's not as simple as just creating a new key.

I did *not* try the new key with the 2.5 API to check that behavior. It sounds like from the above info any newly generated key is limited? It's like the older keys are grandfathered in and still work with the 2.5 API?

tlyu, 2022-09-06:
> I did _not_ try the new key with the 2.5 API to check that behavior. It sounds like from the above info any newly generated key is limited? It's like the older keys are grandfathered in and still work with the 2.5 API?

I created a new OpenWeatherMap account on 2022-09-03, and confirmed that it was unauthorized for the 2.5 OneCall API endpoint that appears in the example code. I also confirmed that at least some of the APIs that are advertised as available to free accounts did work with that API key. I can probably produce protocol traces if you want.

tlyu, 2022-09-06:
These are both using the same new API key that I mentioned earlier. Traces are from `curl -v`.

Failure with 2.5 OneCall API:

```
> GET /data/2.5/onecall?exclude=minutely,hourly,alerts&lat=44.9772995&lon=-93.2654692&appid=REDACTED HTTP/1.1
> Host: api.openweathermap.org
> User-Agent: curl/7.64.1
> Accept: */*
>
< HTTP/1.1 401 Unauthorized
< Server: openresty
< Date: Tue, 06 Sep 2022 17:56:50 GMT
< Content-Type: application/json; charset=utf-8
< Content-Length: 107
< Connection: keep-alive
< X-Cache-Key: /data/2.5/onecall?exclude=minutely,hourly,alerts&lat=44.98&lon=-93.27
< Access-Control-Allow-Origin: *
< Access-Control-Allow-Credentials: true
< Access-Control-Allow-Methods: GET, POST
<
* Connection #0 to host api.openweathermap.org left intact
{"cod":401, "message": "Invalid API key. Please see http://openweathermap.org/faq#error401 for more info."}* Closing connection 0
```

Success with 2.5 Current Weather API:

```
> GET /data/2.5/weather?lat=44.9772995&lon=-93.2654692&appid=REDACTED HTTP/1.1
> Host: api.openweathermap.org
> User-Agent: curl/7.64.1
> Accept: */*
>
< HTTP/1.1 200 OK
< Server: openresty
< Date: Tue, 06 Sep 2022 17:57:26 GMT
< Content-Type: application/json; charset=utf-8
< Content-Length: 478
< Connection: keep-alive
< X-Cache-Key: /data/2.5/weather?lat=44.98&lon=-93.27
< Access-Control-Allow-Origin: *
< Access-Control-Allow-Credentials: true
< Access-Control-Allow-Methods: GET, POST
<
* Connection #0 to host api.openweathermap.org left intact
{"coord":{"lon":-93.2651,"lat":44.9824},"weather":[{"id":800,"main":"Clear","description":"clear sky","icon":"01d"}],"base":"stations","main":{"temp":299.49,"feels_like":299.49,"temp_min":297.91,"temp_max":301.15,"pressure":1019,"humidity":69},"visibility":10000,"wind":{"speed":3.09,"deg":130},"clouds":{"all":0},"dt":1662486447,"sys":{"type":2,"id":2009535,"country":"US","sunrise":1662464474,"sunset":1662511318},"timezone":-18000,"id":5037649,"name":"Minneapolis","cod":200}* Closing connection 0
```

tlyu, 2022-09-06:
> It's like the older keys are grandfathered in and still work with the 2.5 API?

From observations reported by you and others, I can infer that older free API keys are (unofficially?) allowed to continue working with the 2.5 OneCall API, while new free API keys do not work. It also appears that the 2.5 OneCall API is no longer officially documented, at least not anywhere that I can easily find.

caternuson, 2022-09-06:
Thanks for the additional testing and confirming of behavior.

It sounds like this will need to be looked at in more detail. Will need to figure out what exactly is available for free with current keys and API and see if guide can still be supported as shown, or will need to be reworked, etc.

caternuson, 2022-10-03:
This PR adds a new version of the code that should work with free API keys:
https://github.com/adafruit/Adafruit_Learning_System_Guides/pull/2290

Try using the version under the `forecast` subfolder.

caternuson, 2022-11-01:
Closing. Hopefully fixed for now until API access changes again.

closed 2022-11-01 by caternuson