adafruit/Adafruit_Learning_System_Guides #2335: ulab.scipy.signal import spectrogram stopped working with 8.0.0-beta.4

somenice, 2022-11-20:


jepler, 2022-11-20:
I've transferred this report to Learn.

ulab 5.0 (first in CircuitPython 8) made an incompatible change, so that `spectrogram` is in `ulab.utils` rather than `ulab.scipy.signal`. The following examples need to be updated, and the guide text needs to be reviewed in case it should be updated as well:
```
EyeLights_Audio_Spectrum/EyeLights_Audio_Spectrum_CircuitPython/code.py:from ulab.scipy.signal import spectrogram
Feather_Sense_Audio_Visualizer_13x9_RGB_LED_Matrix/audio_spectrum_lightshow/code.py:from ulab.scipy.signal import spectrogram
Feather_Sense_Audio_Visualizer_13x9_RGB_LED_Matrix/waterfall_visualizer/code.py:from ulab.scipy.signal import spectrogram
Ukulele/code.py:from ulab.scipy.signal import spectrogram
ulab_Crunch_Numbers_Fast/waterfall/code.py:from ulab.scipy.signal import spectrogram
```

During  the transitional period, where it's reasonable to accomodate using 7.3 and 8.x, I think the code should read
```
try:
    from ulab.utils import spectrogram
except ImpotError:
    from ulab.scipy.signal import spectrogram
```
then the example should be tested using both 7.3 and 8.0.

jepler, 2022-11-20:
@kattni is this one that eva could potentially take on?

somenice, 2022-11-20:
Thank you for the quick fix @jepler 
I can confirm `from ulab.utils import spectrogram` works as intended.

ps. note the typo in code above `except ImpotError:`
thanks again

jepler, 2022-11-20:
Thanks both for testing and noticing the typo!

kattni, 2022-11-21:
@evaherrada Please add this to your list. You need to update the ulab utils import to the following: 
```
try:
    from ulab.utils import spectrogram
except ImportError:
    from ulab.scipy.signal import spectrogram
```
If you have any questions, please reach out to @jepler in this thread. Thank you!

CedarGroveStudios, 2023-12-31:
The code for the learning guide [Adafruit EyeLights LED Glasses and Driver](https://learn.adafruit.com/adafruit-eyelights-led-glasses-and-driver)  Music-Reactive Lights also needs to be updated.

PR #2336 (2024-01-20): fixed ulab spectrogram imports
Fixes #2335
closed 2024-01-20 by jepler